FreeCAD/FreeCAD #19528: PD Draft can't use sketch as neutral plane. And can't use origin planes either.

PaddleStroke, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

PD Draft can't use sketch as neutral plane. And can't use origin planes either.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ziad-ashraf7, 2025-02-15:
Hi There, Can you please provide me with some details so i can help you on this issue ? 
How to cause this issue in steps ? and what is the expected outcome that you want ?
Hope i can help you in this issue after providing me with some details :)

PaddleStroke, 2025-02-15:
Hi Ziad,
For this issue, create a simple cube with part design.
Add a sketch on the XY plane.
Then try to make a draft operation (the same operation that you worked on
in your last PR). Click on the 'Neutral plane' button. This will start
selection mode for the neutral plane.
Now if you try to select one face of the cube it's OK. If you try to select
the Body's origin's XY plane, it's OK. But you cannot select the sketch
that you created just before. It should be possible to select a sketch
since a sketch has a placement, it's the same as a plane or a face.

So the feature request is to be able to select the sketch.

ziad-ashraf7, 2025-02-16:
I see , so you're trying to select the XY Plane and it works , like this 

[image]

and if you've selected the face it works fine too

[image]

but you just can select the sketch edge not the entire edge , something like that 

[image]

am i understanding the issue corectly ? please let me know if any something that i miss .

ziad-ashraf7, 2025-02-16:
Hi @PaddleStroke , 
Just in case of understanding the issue and start working on it, I just wanted to ask you to answer on my above question to make sure that i can completely understand the issue :)

ziad-ashraf7, 2025-02-16:
No problem , Thank you for you response :)
I think I'm starting to get it, so i need to edit the selecting function to make it able to accept the sketch as a natural plane. 
I'll give it a try to fix it as possible as i can , so can you assign this task for me ?

ziad-ashraf7, 2025-02-17:
Sorry for being late to open the PR , but i have some build issues after pulling the latest changes of the repo , every time i run 
`mkdir build
cd build
cmake ../
make -j$(nproc --ignore=2)`

on the repo directory , i get these errors , can't even catch the problem 

[image]

even after re-cloning the repo i still face the problem after making the initial build and editing some piece of code and re-building the project

FlachyJoe, 2025-02-17:
@ziad-ashraf7 You can open the PR even if it doesn't build or simply push your modification in your github repo so we will give a look to your code.

ziad-ashraf7, 2025-02-17:
@FlachyJoe okay but at first i want to make sure that my code modifcation is correct , so i will give it a second try now and build the project , and if it failed , I'll open the PR

ziad-ashraf7, 2025-02-17:
@FlachyJoe I've created the PR without building and testing my changes and trying to add the sketch to the neural plane.

ziad-ashraf7, 2025-02-19:
Hi @FlachyJoe ,
I've solved the build problems that i had before and went to test my modifications , but i just wanted to ask you about the log messages that i put in my code:
when i want to test my work i wanted to add some log messages like this 
`// Reversed pull direction
    bool reversed = Reversed.getValue();
    if (reversed)
    {
        std::cout<<"REVERSED\n";
        angle *= -1.0;
    }`

or this 

`if (refPlane->isDerivedFrom<PartDesign::Plane>()) {
            std::cout<<"Neutral Plane is a Plane\n";
            PartDesign::Plane* plane = static_cast<PartDesign::Plane*>(refPlane);
            Base::Vector3d b = plane->getBasePoint();
            Base::Vector3d n = plane->getNormal();
            neutralPlane = gp_Pln(gp_Pnt(b.x, b.y, b.z), gp_Dir(n.x, n.y, n.z));
        }`

and i can make sure that these pieces of code is excuted while running the project from the terminal. but the truth is i can't see any log message on the terminal when i exc this pieces. am i wrong about using `cout` to make some log messages ?
or am i writing it in the wrong place ?

ziad-ashraf7, 2025-02-20:
Hi @FlachyJoe @PaddleStroke , 
Just wanted to make sure , is this result is Accepted ? and how to make sure that the neutral plane for the sketch is acting as expected ?

https://github.com/user-attachments/assets/84f31b1e-5b12-4fe9-b99a-481f6c00aec5

PaddleStroke, 2025-02-21:
In your video we only see that it is possible to select the sketch. But it does not show if the draft is actually working properly then. Can you confirm that the draft opération works as expected?

PaddleStroke, 2025-02-21:
That looks good!

ziad-ashraf7, 2025-02-21:
okay so should i open the PR ?

PaddleStroke, 2025-02-21:
Yes

ziad-ashraf7, 2025-03-03:
Hi @maxwxyz , I've created a PR here #19762 and need some review for it.