FreeCAD/FreeCAD #19530: BIM: error creating a wall: module 'numpy.typing' has no attribute 'NDArray'

furgo16, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD
2. Switch to BIM workbench
3. Click on the Arch_Wall command
4. Click on a point in the 3D window to start the wall
5. Click on a point in the 3D window to end the wall
6. Result: the wall is created, but this error is returned on the report window:

```python
11:36:36  pyException: Traceback (most recent call last):
  File "/snap/freecad/1294/usr/Mod/BIM/nativeifc/ifc_observer.py", line 127, in slotRemoveDynamicProperty
    from nativeifc import ifc_psets
  File "/snap/freecad/1294/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/snap/freecad/1294/usr/Mod/BIM/nativeifc/ifc_psets.py", line 29, in <module>
    from nativeifc import ifc_tools
  File "/snap/freecad/1294/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/snap/freecad/1294/usr/Mod/BIM/nativeifc/ifc_tools.py", line 39, in <module>
    from ifcopenshell.util import element
  File "/snap/freecad/1294/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/snap/freecad/1294/lib/python3.10/site-packages/ifcopenshell/util/element.py", line 22, in <module>
    import ifcopenshell.util.representation
  File "/snap/freecad/1294/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/snap/freecad/1294/lib/python3.10/site-packages/ifcopenshell/util/representation.py", line 22, in <module>
    import ifcopenshell.util.placement
  File "/snap/freecad/1294/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/snap/freecad/1294/lib/python3.10/site-packages/ifcopenshell/util/placement.py", line 25, in <module>
    MatrixType = npt.NDArray[np.float64]
<class 'AttributeError'>: module 'numpy.typing' has no attribute 'NDArray'
```

I can reproduce this both with a source build and with the daily build snap package. What they both have in common is that I'm using the latest `ifcopenshell` from pip. It used to be `0.8.0` until recently, when it switched to `0.8.1`. I think ifcopenshell `0.8.1` is what is causing the issue.  

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40128 (Git) Snap 1294
Build type: Release
Branch: main
Hash: f74a44ab4f92813d80902d8af30a41fb9ee25075
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical/physical DPI: 96/159.34
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-02-11:
IfcOpenShell 0.8.1 was released on PyPi two days ago: https://pypi.org/project/ifcopenshell/#history

Syres916, 2025-02-11:
It would appear to be how FreeCAD is packaged, I too can create a Wall but get a completely different error message:

```
11:19:50  pyException: Traceback (most recent call last):
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/Mod/BIM/nativeifc/ifc_observer.py", line 135, in slotRemoveDynamicProperty
    from nativeifc import ifc_psets
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/Mod/BIM/nativeifc/ifc_psets.py", line 29, in <module>
    from nativeifc import ifc_tools
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 56, in <module>
    from nativeifc import ifc_viewproviders
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/Downloads/FC_40176/squashfs-root/usr/Mod/BIM/nativeifc/ifc_viewproviders.py", line 27, in <module>
    translate = FreeCAD.Qt.translate
                ^^^^^^^
<class 'NameError'>: name 'FreeCAD' is not defined
```

When I 'upgraded' to IfcOpenShell 0.8.1 I cleared out the /home/username/.local/share/FreeCAD/AdditionalPythonPackages/py311 folder so all the dependencies would be brought along with IfcOpenShell which include `typing_extensions-4.12.2.dist-info`

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.8.1, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.5
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.7
```

kaiwas, 2025-02-11:
The problem is reproduced not only with walls. If you add a column (or anything from the BIM components) - the console is all red.

[image]

Syres916, 2025-02-11:
PR submitted for the name 'FreeCAD' is not defined: #19533 

@furgo16 here is my contents of the py311 folder:

<pre><font color="#729FCF"><b>bin</b></font>                           <font color="#729FCF"><b>numpy.libs</b></font>
<font color="#729FCF"><b>dateutil</b></font>                      <font color="#729FCF"><b>__pycache__</b></font>
<font color="#729FCF"><b>ifcopenshell</b></font>                  <font color="#729FCF"><b>python_dateutil-2.9.0.post0.dist-info</b></font>
<font color="#729FCF"><b>ifcopenshell-0.8.1.dist-info</b></font>  <font color="#729FCF"><b>shapely</b></font>
<font color="#729FCF"><b>isodate</b></font>                       <font color="#729FCF"><b>shapely-2.0.7.dist-info</b></font>
<font color="#729FCF"><b>isodate-0.7.2.dist-info</b></font>       <font color="#729FCF"><b>shapely.libs</b></font>
<font color="#729FCF"><b>lark</b></font>                          <font color="#729FCF"><b>six-1.17.0.dist-info</b></font>
<font color="#729FCF"><b>lark-1.2.2.dist-info</b></font>          six.py
<font color="#729FCF"><b>numpy</b></font>                         <font color="#729FCF"><b>typing_extensions-4.12.2.dist-info</b></font>
<font color="#729FCF"><b>numpy-2.2.2.dist-info</b></font>         typing_extensions.py
</pre>

furgo16, 2025-02-11:
Let me report properly on my `py311` folder later on. @Syres916 out of interest, how do did you install FreeCAD? With a conda source build? The reason I'm asking is because I thought (at least on the AppImage) we were shipping ifcopenshell 0.7.0. See https://github.com/FreeCAD/FreeCAD/issues/19051#issuecomment-2649221339 for the context.

Syres916, 2025-02-11:
@furgo16 it's an extracted AppImage in which I was helping forum user, see https://forum.freecad.org/viewtopic.php?p=810304&sid=1b47af5a83d8c67767ae682d01f250db#p810304

Syres916, 2025-02-11:
@furgo16 so when I look down your _py310/site-packages_ unless I'm not mistaken `numpy` is missing and is possibly held elsewhere e.g. _dist-packages_ (I've never been near a Snap package)?? If the version of `numpy` is not being updated alongside `IfcOpenShell` I would imagine things can get out of whack.

furgo16, 2025-02-11:
Good hint on `dist-packages`, thanks.

On the source build: I upgraded again to 0.8.1 manually (`pip install ifcopenshell -U`) and I cannot reproduce the issue anymore. There, numpy is provided not by pip, but by the `python3-numpy` .deb package that I have installed (numpy version 1.21.5). BUT, it seems I have another numpy that takes precedence, which is installed at `$HOME/.local/lib/python3.10/site-packages/` and reports a higher version: `1.26.4`. This was probably a pip user install.

On the snap build: same situation, but there the numpy version from `python3-numpy` (1.21.5) cannot be overriden by the user installing a newer numpy package.

My guess is that it's how the pip dependencies are listed upstream. Probably ifcopenshell 0.8.0 worked with numpy <= 1.21.5 and ifcopenshell 0.8.1 works with numpy > 1.21.5. So if the 0.8.1 pip package does not specify a specific numpy version, either the latest numpy will be installed as a dependency, or a previous numpy will be kept installed if present. In the later case, if ifcopenshell and that previous numpy are not compatible, the error will occur. At least that's my theory.

furgo16, 2025-02-11:
Something else I've observed. This is with the `python3-numpy` .deb package installed:

On the terminal:
```python
>>> import numpy
>>> numpy.__version__
'1.21.5'
>>> import numpy.typing
```

Inside FreeCAD installed from the snap package:

```python
>>> import numpy
>>> numpy.__version__
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: module 'numpy' has no attribute '__version__'
>>> 
```

It would appear like numpy is not properly loaded, but I can't figure out why.

furgo16, 2025-02-12:
> It would appear like numpy is not properly loaded, but I can't figure out why.

For some reason, all the files in the numpy installation location are empty, only (empty) directories are left.

```
ls /snap/freecad/current/usr/lib/python3/dist-packages/numpy
compat  distutils  f2py  lib     ma         polynomial  testing  typing
core    doc        fft   linalg  matrixlib  random      tests
```

It's not only causing issues with the BIM workbench, but also with the Add-on manager, which uses numpy as well.

This looks like an issue specific to the snap package, then. @maxwxyz @luzpaz could you please transfer this issue to the https://github.com/FreeCAD/FreeCAD-snap repository?

maxwxyz, 2025-02-12:
I have no permissions there @luzpaz could you please?

closed 2025-02-13 by chennes