FreeCAD/FreeCAD #19532: PD Draft does not accept negative angle.

PaddleStroke, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

PD Draft does not accept negative angle.
A negative angle should just do the same as checking the 'reverse pull direction'.

Why not just using the checkbox you ask? Well if you want to control the angle from a property and be able to change it from a copy on change link. Now it's not possible.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ziad-ashraf7, 2025-02-13:
Hi there, 
I'm interested about this issue , can you just give some context about ? 
How to cause the issue on the app by steps ? 
what is the current behavior and the expected one that you want to have ?
i hope to help you in this issue after having a good context from you !

PaddleStroke, 2025-02-13:
[draft.zip](https://github.com/user-attachments/files/18781155/draft.zip)
Thanks for jumping in ! 
Here's a file. Open it, double click on the draft, try to enter a negative value for the angle. It's not possible. Currently you need to click the reverse checkbox if you want to reverse the angle.
What would be better is that entering a negative angle is considered the same as checking the reverse checkbox.

It does not mean that we should remove the checkbox. I think it should stay there and still reverse correctly. So (negative angle + checked box) should do the same as (positive angle + uncheck box)

Let me know when you have a PR up and I can review it.

FlachyJoe, 2025-02-13:
Quick workaround:
```python
obj = App.ActiveDocument.Draft
obj.addProperty("App::PropertyAngle", "TrueAngle")
obj.setExpression("Angle", "abs(TrueAngle)")
obj.setExpression("Reversed", "TrueAngle>0?False:True")

obj.TrueAngle = "-20°"
obj.recompute()
```

PaddleStroke, 2025-02-13:
@ziad-ashraf7 you can find all the instructions in https://wiki.freecad.org/Compiling

ziad-ashraf7, 2025-02-13:
Hi @PaddleStroke , 
Just wanted to ask you , is this the result you are looking for ?

https://github.com/user-attachments/assets/5cfefd05-eb91-46d1-95bb-6d323eaf4f87

PaddleStroke, 2025-02-13:
@ziad-ashraf7 yes exactly. Great work, and very fast too. You sure setup the IDE and did the change quite fast ! :)
Now you need to open a pull request to freecad's main repo.
Thanks

ziad-ashraf7, 2025-02-13:
Thank you @PaddleStroke !
should i run some tests before opening the PR ?

FlachyJoe, 2025-02-13:
> should i run some tests before opening the PR ? 
You can run `FreeCAD -t TestPartDesignApp` but you can also let the CI do the job. The PR can be created as draft then the checks are triggered automatically and if they success you will mark the PR ready for review.

> also what about the `build` directory that i made for the build process , should i submit the PR with it ?
No, only the modified code is needed. 

Are you ok with git? Feel free to ask for help.

ziad-ashraf7, 2025-02-13:
@FlachyJoe Thanks for Clarifying this to me !
I've created the PR here https://github.com/FreeCAD/FreeCAD/pull/19584
Hope everything works fine

PR #19584 (2025-02-14): PartDesign: Allow Draft negative angles
Fix #19532
Changes staged in:
src/Mod/PartDesign/App/FeatureDraft.cpp

Modified the `Constrains`'s `LowerBound` to accept the negative angles
closed 2025-02-14 by chennes