FreeCAD/FreeCAD #19534: Assembly needs array tools

PaddleStroke, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

Assembly needs array tools. Currently one can use link group, but then the position of each link is independant.
One can use draft array tools. But this requires an annoying change of workbench. Also the tool could be adapted to better suit assembly workflow.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-02-11:
The BOM tool would need to understand those arrays in order to give the right number of instances.

kadet1090, 2025-02-11:
Can't we simply put draft tools into assembly? We should avoid duplicating tools.

pierreporte, 2025-02-11:
Maybe move those tools into core, like for new datum?

PaddleStroke, 2025-02-11:
I don't think this can be in core because it needs OCC I think.

First, without talking about assembly, I think the array tools should be in Part not Draft. It doesn't make much sense that they are in draft don't you think?

Secondly, if we go through merging part and partdesign, then the logical place for this would rather be assembly than partdesign I think. No?

I agree that duplicating is not a good idea.

kadet1090, 2025-02-11:
Part seems like a good place, but merging the two workbenches is a long way ahead still. Draft is built-in workbench, it provides working tool that should cover this usecase. Later we may try provide another tool that will deprecate one from both part design and draft.

TimBuildIt, 2025-03-12:
assembly array would be awesome

TimBuildIt, 2025-03-12:
it would be nice if you could do a linear array with control over the distance between each element. for example if you wanted 3 copies different intervals from the origional

pierreporte, 2025-03-12:
Yes it would be very useful to manually set some intervals manually, while other remain with the given default value. But it need to be input in the 3D view to be comfortable to use.

belinski-roman, 2025-03-13:
> Разве мы не можем просто добавить чертёжные инструменты в сборку? Нам следует избегать дублирования инструментов.

When using an array from Draft, the nesting structure in the tree will be reversed.