FreeCAD/FreeCAD #19538: Padding of a master sketch does not work

greenscreenflicker, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have the following master sketch:

[image]

I want to extrude/pad the orange marked section. Cannot do it.

The same for the orange marked area

[image]

The files for experimenting yourself are attached.

[halter_schwamm_1a.zip](https://github.com/user-attachments/files/18752024/halter_schwamm_1a.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-02-11:
Use [Sketcher_Split](https://wiki.freecad.org/Sketcher_Split) at the T intersections so you can select one part or the other of the big edges.

maxwxyz, 2025-02-12:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12820

closed 2025-02-12 by maxwxyz