FreeCAD/FreeCAD #19541: BIM: IfcOpenShell is not upgraded after 'IfcOpenShell update' is run on FC 1.0

semhustej, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce on FreeCAD 1.0.0:

- start `Utils -> IfcOpenShell update`
- Click OK in the dialogue informing about installing version 0.8.1
- The upgrade seems to happen (takes some time and downloading). At the end the user is informed, that IfcOpenShell has been updated
- Restart FreeCAD
- run `import ifcopenshell` in python console
- run `print(ifcopenshell.version)` in python console
- 0.0.0 is returned
- start start `Utils -> IfcOpenShell update`
- dialogue informing about installing version 0.8.1 shows up

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-11:
This topic is probably somehow related
https://github.com/FreeCAD/FreeCAD/issues/19530

yorikvanhavre, 2025-02-12:
Probably related to #19540