FreeCAD/FreeCAD #19545: Part Design - Crash when deleting Binder after deleting Binder source

steveklett, 2025-02-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was cleaning up some older sketches and features in a project and deleting some things. I deleted a sketch and a pocket from 1 body, then noticed an error from a Binder in another body. I then found that Binder and when you delete it from the tree FC crashes immediately.

There are warnings about dependencies but I usually ignore those and delete anyway, then react to the next error and delete the next item. This normally works fine and is faster than chasing the dependency warnings in complex projects.

**Reproduction Steps:**

1. Open the attached file
2. Select Partition Dado (Pocket005) and it's Sketch (Sketch010)
[image]

3. Press DEL to delete them
4. Note the error in the Report view: "AdditiveLoft003: Failed to get shape of Profile Binder009.?Face1"
5. Search for Binder009 in the tree
6. Select it and press DEL - Instant crash
[image]

[issue - delete binder causes crash.FCStd.tar.gz](https://github.com/user-attachments/files/18760024/issue.-.delete.binder.causes.crash.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-02-12:
@AIRCAP, the crash is caused by `nullptr` dereference made in #16791 (_obj_ is `nullptr`). Could you take a look at this crash?

https://github.com/FreeCAD/FreeCAD/blob/8899e3476e770f3d1da5099683a4a167b21fd0b1/src/Mod/PartDesign/App/FeatureLoft.cpp#L78

xtemp09, 2025-02-12:
@AIRCAP, if you take a look at the "old code", it checks for `nullptr`, while the new does not.

maxwxyz, 2025-02-14:
@xtemp09 he is inactive for some time now, are you able to fix it?

closed 2025-07-18 by chennes