FreeCAD/FreeCAD #19556: Add-On Manager: cannot install add-ons

furgo16, 2025-02-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Steps to reproduce

1. Start Add-On Manager
2. Select a workbench to install
3. Press the <kbd>Install</kbd> button
4. Install does not happen. The error below is shown:

```python
14:10:55  propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.1872\" (uid=1000 pid=1688015 comm=\"/snap/freecad/1295/usr/bin/FreeCAD - --single-inst\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2257 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
14:10:55  nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.1872\" (uid=1000 pid=1688015 comm=\"/snap/freecad/1295/usr/bin/FreeCAD - --single-inst\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2257 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
14:10:55  "Object path cannot be empty"
14:11:37  pip timed out
14:11:46  Traceback (most recent call last):
14:11:46    File "/snap/freecad/1295/usr/Mod/AddonManager/AddonManager.py", line 827, in launch_installer_gui
14:11:46      self.installer_gui.run()  # Does not block
14:11:46    File "/snap/freecad/1295/usr/Mod/AddonManager/addonmanager_installer_gui.py", line 106, in run
14:11:46      deps = MissingDependencies(self.addon_to_install, self.addons)
14:11:46    File "/snap/freecad/1295/usr/Mod/AddonManager/Addon.py", line 835, in __init__
14:11:46      __import__(py_dep)
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
14:11:46      return original_import(name, *args, **kwargs)
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/numba/__init__.py", line 42, in <module>
14:11:46      from numba.np.ufunc import (vectorize, guvectorize, threading_layer,
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
14:11:46      return original_import(name, *args, **kwargs)
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/numba/np/ufunc/__init__.py", line 3, in <module>
14:11:46      from numba.np.ufunc.decorators import Vectorize, GUVectorize, vectorize, guvectorize
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
14:11:46      return original_import(name, *args, **kwargs)
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/numba/np/ufunc/decorators.py", line 3, in <module>
14:11:46      from numba.np.ufunc import _internal
14:11:46    File "/snap/freecad/1295/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
14:11:46      return original_import(name, *args, **kwargs)
14:11:46  SystemError: initialization of _internal failed without raising an exception

```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40230 (Git) Snap 1295
Build type: Release
Branch: main
Hash: 72c37f9b7555d175ca6e3dac9855a89dd968ff73
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-14:
I cannot reproduce, seems to be a snap issue.

furgo16, 2025-02-14:
Today's daily build is giving me other errors. It freezes while checking for updates:

<details>

```python
freecad
FreeCAD 1.1.0, Libs: 1.1.0devR40239 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: $HOME/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1299/lib/python3.10/site-packages:/snap/freecad/1299/usr/lib/python3/dist-packages
Gtk-Message: 19:48:51.417: Failed to load module "canberra-gtk-module"
Gtk-Message: 19:48:51.418: Failed to load module "canberra-gtk-module"
Qt: Session management error: Could not open network socket
propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.85873\" (uid=1000 pid=2430214 comm=\"/snap/freecad/1299/usr/bin/FreeCAD\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2694 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.85873\" (uid=1000 pid=2430214 comm=\"/snap/freecad/1299/usr/bin/FreeCAD\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=2694 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
"Object path cannot be empty"
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
  File "/snap/freecad/1299/usr/Mod/AddonManager/Widgets/addonmanager_widget_progress_bar.py", line 65, in __init__
    raise ValueError(
ValueError: Current task must be between 0 and the number of tasks (1), not 3
Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/AddonManager.py", line 872, in update_progress_bar
    progress = Progress(
[... 465 lines cut ...]

Traceback (most recent call last):
  File "/snap/freecad/1299/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 555, in run
    checker.check_workbench(repo)
  File "/snap/freecad/1299/usr/Mod/AddonManager/addonmanager_workers_startup.py", line 590, in check_workbench
    self.git_manager.repair(wb.url, clonedir)
  File "/snap/freecad/1299/usr/Mod/AddonManager/addonmanager_git.py", line 271, in repair
    raise e
[... cut ...]
  File "/snap/freecad/1299/usr/Mod/AddonManager/addonmanager_git.py", line 457, in _synchronous_call_git
    raise GitFailed(
addonmanager_git.GitFailed: Git returned a non-zero exit status: -1
Called with: /snap/freecad/1299/usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-library $HOME/snap/freecad/common/Mod/parts_library

Returned stderr:
Cloning into '$HOME/snap/freecad/common/Mod/parts_library'...
fatal: early EOF
```

</details>

Eventually I got Add-On Manager to not crash after FreeCAD crashed and restarted.

furgo16, 2025-02-17:
The `pip timed out` exception may or may not be what triggers the issue, but ig might be related:

https://github.com/FreeCAD/FreeCAD/blob/04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd/src/Mod/AddonManager/addonmanager_python_deps_gui.py#L132-L135

This is I believe because FreeCAD is trying to call an external executable (`freecad.pip`) from within the confined environment:

https://github.com/FreeCAD/FreeCAD/blob/04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd/src/Mod/AddonManager/addonmanager_python_deps_gui.py#L132-L135

The `call_args` below will result in an error:

https://github.com/FreeCAD/FreeCAD/blob/04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd/src/Mod/AddonManager/addonmanager_utilities.py#L559-L562

chennes, 2025-02-18:
So it appears that this `freecad.pip` is not the correct way to handle pip on snaps either. Unfortunately my only Linux systems are all on ARM, which we don't provide snaps for, so I can't test anything directly. What *is* the right way to call pip in a snap?

furgo16, 2025-02-20:
> What is the right way to call pip in a snap?

@chennes 

From within a snap, programmatically, I would say through the Python executable included in the snap (`python pip`, as I think it's done for the other platforms in the code).

From outside a snap, as a user, with `freecad.pip`.

There are a couple of caveats:

1. https://github.com/FreeCAD/FreeCAD-snap/issues/68 => the `pip` executable in the snap always runs in user mode (presumably so that `sudo` is not required), so it implicitly adds the `--user` arg. This means that for snaps, the `--target` arg cannot be passed. Python packages in the snap are always installed at `$HOME/snap/freecad/common/.local/lib/python3.10/site-packages`.
2. https://github.com/FreeCAD/FreeCAD-snap/issues/167 => this is a related issue that needs to be resolved. It does not affect the installation per se, but it's worth taking it into account. With the current situation, even if Add-On Manager successfully installs a Python package we ship (i.e. `ifcopenshell`), FreeCAD won't see it.

Point 2 is not critical and needs to be resolved in the snap rather than in Add-On Manager. Point 1 is the one we need to get right.

chennes, 2025-02-20:
OK, so the AM needs to check to see if it's in a snap when it builds the pip command, as we do, but instead of using a unique pip call it needs to drop the `--target` arg.

chennes, 2025-02-21:
Is it possible for you to test the following code block by replacing the function at the end of addonmanager_utilities.py with this version?
```python
def create_pip_call(args: List[str]) -> List[str]:
    """Choose the correct mechanism for calling pip on each platform. It currently supports
    either `python -m pip` (most environments) or `pip` (Snap packages). Returns a list
    of arguments suitable for passing directly to subprocess.Popen and related functions."""
    snap_package = os.getenv("SNAP_REVISION")
    if snap_package:
        call_args = ["pip", "--disable-pip-version-check"]
        call_args.extend(args)
        # The Snap pip automatically adds the --user option, which is not compatible with the
        # --target option, so we have to remove --target and its argument, if present
        index = call_args.index("--target")
        if index:
            del call_args[index:index+1]  # The --target option and its argument
    else:
        python_exe = get_python_exe()
        if not python_exe:
            raise RuntimeError("Could not locate Python executable on this system")
        call_args = [python_exe, "-m", "pip", "--disable-pip-version-check"]
        call_args.extend(args)
    return call_args
```

furgo16, 2025-02-21:
@chennes the content of snaps cannot be modified, so I cannot insert it directly. I'll try to see if I can add it to a macro, though.

If the macro route doesn't work, I'll try to build a new snap from a branch with those changes. It will take me longer, but I'll report back.

chennes, 2025-02-21:
It should work as a macro: try passing it `["install", "--target", "/some/random/path", "gersemi"]` and see if the package installs.

furgo16, 2025-02-21:
@chennes I'm still building the macro, as it needs some other bits and pieces to be copied over. One thing I've noticed though: the snipped above removes only `--target` from the args list. It should remove the path as well.

furgo16, 2025-02-21:
@chennes it worked, with caveats:

1. `create_pip_call` correctly removes the `--target` arg, but it doesn't remove the target path (`some/random/path`)
2. `create_pip_call` fails if the `--target` arg is not passed to it

Here's the macro that I used:

<details>

```python

import FreeCAD
import addonmanager_utilities as utils
# import freecad.utils as utils
import subprocess
import os
from typing import List

try:
    from freecad.utils import get_python_exe
except ImportError:

    def get_python_exe():
        """Use shutil.which to find python executable"""
        return shutil.which("python")

def create_pip_call(args: List[str]) -> List[str]:
    """Choose the correct mechanism for calling pip on each platform. It currently supports
    either `python -m pip` (most environments) or `pip` (Snap packages). Returns a list
    of arguments suitable for passing directly to subprocess.Popen and related functions."""
    snap_package = os.getenv("SNAP_REVISION")
    if snap_package:
        call_args = ["pip", "--disable-pip-version-check"]
        call_args.extend(args)
        # The Snap pip automatically adds the --user option, which is not compatible with the
        # --target option, so we have to remove --target and its argument, if present
        index = call_args.index("--target")
        if index:
            del call_args[index:index+1]  # The --target option and its argument
    else:
        python_exe = get_python_exe()
        if not python_exe:
            raise RuntimeError("Could not locate Python executable on this system")
        call_args = [python_exe, "-m", "pip", "--disable-pip-version-check"]
        call_args.extend(args)
    return call_args

def call_pip(args: List[str]) -> List[str]:
    """Tries to locate the appropriate Python executable and run pip with version checking
    disabled. Fails if Python can't be found or if pip is not installed."""

    try:
        call_args = create_pip_call(args)
    except RuntimeError as exception:
        raise PipFailed() from exception

    try:
        proc = utils.run_interruptable_subprocess(call_args)
    except subprocess.CalledProcessError as exception:
        raise PipFailed("pip timed out") from exception

    if proc.returncode != 0:
        raise PipFailed(proc.stderr)

    data = proc.stdout
    return data.split("\n")

call_args = ["install", "--target", "gersemi"]
output = call_pip(call_args)
print(*output, sep='\n')

call_args = ["list", "--target", "-v"]
output = call_pip(call_args)
print(*output, sep='\n')

```

</details>

And here's the output:

```
Collecting gersemi
  Using cached gersemi-0.19.1-py3-none-any.whl.metadata (30 kB)
Requirement already satisfied: appdirs in /snap/freecad/1305/usr/lib/python3/dist-packages (from gersemi) (1.4.4)
Requirement already satisfied: lark>=1.0 in /snap/freecad/1305/lib/python3.10/site-packages (from gersemi) (1.2.2)
Requirement already satisfied: pyyaml<7,>=5 in /snap/freecad/1305/usr/lib/python3/dist-packages (from gersemi) (5.4.1)
Using cached gersemi-0.19.1-py3-none-any.whl (80 kB)
Installing collected packages: gersemi
Successfully installed gersemi-0.19.1

Package Version Location                                                          Installer
------- ------- ----------------------------------------------------------------- ---------
gersemi 0.19.1  /home/dpm/snap/freecad/common/.local/lib/python3.10/site-packages pip
numpy   1.26.4  /home/dpm/snap/freecad/common/.local/lib/python3.10/site-packages pip
```

chennes, 2025-02-21:
OK, thanks: an off-by-one-error and a failure-to-RTFM error... try this one?
```
def remove_target_option(args: List[str]) -> List[str]:
    # The Snap pip automatically adds the --user option, which is not compatible with the
    # --target option, so we have to remove --target and its argument, if present
    try:
        index = args.index("--target")
        del args[index:index+2]  # The --target option and its argument
    except ValueError:
        pass
    return args

def create_pip_call(args: List[str]) -> List[str]:
    """Choose the correct mechanism for calling pip on each platform. It currently supports
    either `python -m pip` (most environments) or `pip` (Snap packages). Returns a list
    of arguments suitable for passing directly to subprocess.Popen and related functions."""
    snap_package = os.getenv("SNAP_REVISION")
    appimage = os.getenv("APPIMAGE")
    if snap_package:
        args = remove_target_option(args)
        call_args = ["pip", "--disable-pip-version-check"]
        call_args.extend(args)
    elif appimage:
        args = remove_target_option(args)
        python_exe = fci.DataPaths.home_dir + "bin/python"
        call_args = [python_exe, "-m", "pip", "--disable-pip-version-check"]
        call_args.extend(args)
    else:
        python_exe = get_python_exe()
        if not python_exe:
            raise RuntimeError("Could not locate Python executable on this system")
        call_args = [python_exe, "-m", "pip", "--disable-pip-version-check"]
        call_args.extend(args)
    return call_args
```

chennes, 2025-02-21:
Should be fixed by https://github.com/FreeCAD/FreeCAD/pull/19766