FreeCAD/FreeCAD #19558: Rotulos de vista no aparecen cuando se exporta a svg, pdf o dxf

jsalazark, 2025-02-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Cuando exporto un dibujo con TechDraw, en el archivo resultante no aparecen los rotulos de vistas , ni de detalles.
Se podria mejorar o corregir esto.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Mexico (es_MX)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.5
  * Estimate 0.1.2
  * FrameForge 0.1.2
  * freecad.gears 1.3.0
  * Quetzal
```

### Subproject(s) affected?

None

### Anything else?

[Tobera 45 PH #4.pdf](https://github.com/user-attachments/files/18769481/Tobera.45.PH.4.pdf)
[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-14:
"When I export a drawing with TechDraw, the resulting file does not show the view or detail labels.
Could this be improved or corrected."

WandererFan, 2025-02-14:
That is the expected behaviour.  The labels are only an aid during construction of the drawing.  If you want text that stays visible on export/print, you should use the "Caption" property.

Translation courtesy DeepL.

Ese es el comportamiento esperado.  Las etiquetas son sólo una ayuda durante la construcción del dibujo.  Si desea que el texto permanezca visible al exportar/imprimir, debe utilizar la propiedad «Caption».

Traducción por cortesía de DeepL.

maxwxyz, 2025-02-14:
Closing then.

closed 2025-02-14 by maxwxyz