FreeCAD/FreeCAD #19559: BIM_Door creates extra hole on antoher walls

ErtugrulEge, 2025-02-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Sometimes doors make extra holes on another walls. According to my observations, the problem occurs mostly if the distance of the wall is lower than the length of the door.Current distance in the sketch is 189cm (in attached file).Please test the distance for example make it 150cm or 300 cm and see the difference.

Solution is in fact simple, changing Hole Depth property to wall thickness (for example my walls are 20cm) solves the problem but Whenever we add a door we have to manually change this property.

As mentioned in the from "Bim.Door.Hole_Depth = Host Body.width" can be an autonmatic solution for this

[image]

[image]

[door_hole_test.zip](https://github.com/user-attachments/files/18769280/door_hole_test.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

BIM

### Anything else?

Forum topic: https://forum.freecad.org/viewtopic.php?t=92360

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-02-12:
Yes, this is in fact deliberated in details earlier, hope has a gap to solve that :)

paullee0, 2025-02-13:
Previous discussion - https://forum.freecad.org/viewtopic.php?style=1&p=362890#p362890

ErtugrulEge, 2025-02-13:
I am just a simple user.If I want to open a door on a wall, I don't want another hole on another wall. I think, this should be a bug as a user.

kaiwas, 2025-02-13:
> I am just a simple user.

If you look at it from the user's point of view, the possibility of a window existing outside the wall also looks strange. The only time this might be necessary is during the process of creating a window.

edragos, 2025-02-13:
Hello!
The problem is old. It also appears in FreeCad 0.20.
If the height of the door is greater than the width of the wall, a hole appears in the opposite wall.
For example:
A wall with dimensions of 3000mm X 2000mm.
If we put a door with a height of 2100mm on the 3000mm side of the wall, an opposite hole appears.
If we change Hole Depth the hole disappears, but Opening the door the result is wrong.
If we reduce the height of the door, or increase the width of the wall, the hole disappears.

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Europe (en_150)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ArchTextures
  * ExplodedAssembly
  * fasteners 0.5.37

All the best to the FreeCad Community!

paullee0, 2025-02-14:
@maxwxyz IMHO we can leave this open until it is improved :)

maxwxyz, 2025-02-14:
Not sure what the solution would be? If the default is changed, there will be an issue that the hole does not go through the entire wall object.

paullee0, 2025-02-15:
I revisited the previous FC Discussion and has some more thoughts as outlined below (maybe for myself?):-

1.  When processSubShapes() use getSubVolume of its Children (Window/Door, Roof etc.) to obtain the subtraction volume, it pass which object/Wall is calling
2.  Now getSubVolume can study the Wall, thickness, shape whatever, and construct the appropriate sub-volume to subtract - may just start with thickness for simplicity

Maybe nothing needs changes in ArchWall, but at ArchComponent and ArchWindow.

ErtugrulEge, 2025-02-15:
I don't know about complex walls, but I need(I think, most of the users also) plain and simple rectangular prism shaped walls and it seems like the problem can be solved at least for simple walls.

yorikvanhavre, 2025-02-18:
My 2 cents here:

- If the window creates opening(s) in a wall that is a complex wall (sketch-based for ex), there is fundamentally no way to make sure which parts should be cut and which not.
- The current default cut depth is one meter. This is to make sure we cut through all of a wall. This should in theory be good as it's more than 99% of the wall thicknesses out there (okay, besides medieval walls maybe 😅 ) and more than 99% of the rooms. It's not a bad default.
- Making it dependent on the wall thickness, like @paullee0 suggests on the forum, is interesting. Maybe something like this: If the depth is zero, then it adopts the wall thickness + a margin (10%?)

paullee0, 2025-02-18:
> My 2 cents here:
> 
> * If the window creates opening(s) in a wall that is a complex wall (sketch-based for ex), there is fundamentally no way to make sure which parts should be cut and which not.
....
> * Making it dependent on the wall thickness, like [@paullee0](https://github.com/paullee0) suggests on the forum, is interesting. Maybe something like this: If the depth is zero, then it adopts the wall thickness + a margin (10%?)

In fact, there is a way.

With ArchSketch, on which wall segment an ArchObject is attached to is declared by user and saved. 

Then, the extrusion of each wall segment could be done  per segment.  We could just punch hole in the exact wall segment before fusing them all.  There is no need to care about each wall segement thickness with this approach.

Wall thickness + margin is the current 'quick fix' which still have slight chance of punching in adjacent wall.

yorikvanhavre, 2025-02-20:
Let's reopen the issue and try to work on something, @paullee0 ?

paullee0, 2025-02-20:
> Let's reopen the issue and try to work on something, [@paullee0](https://github.com/paullee0) ? I understand you are preparing something already? 😇

Yes, the draft is in FC forum.  Having a further refactoring of the getSubVolume() in the meantime.

paullee0, 2025-02-20:
Anyone would like to further test?  (FC forum is down now)

[ArchWindow.py.txt](https://github.com/user-attachments/files/18893486/ArchWindow.py.txt)

ffeliziani-tpmc, 2025-02-21:
I tested the PR from the forum but I get an error when trying to add a door to a wall:

[image]

am I on the wrong base version of freecad?

semhustej, 2025-02-21:
> Anyone would like to further test? (FC forum is down now)
> 
> [ArchWindow.py.txt](https://github.com/user-attachments/files/18893486/ArchWindow.py.txt)

Tested with ArchComponent.py from the forum and it works fine on my simple test file (as did the version from the forum). Using 1.1dev version.

[image]

paullee0, 2025-02-21:
> am I on the wrong base version of freecad?
> 
> Currently on
> 
> OS: Arch Linux (GNOME/gnome/wayland) Architecture: x86_64 Version: 1.0.0.39109 (Git) Build type: Release Branch: makepkg Hash: [2fcc531](https://github.com/FreeCAD/FreeCAD/commit/2fcc5317fe3aee96ca73475986a577719fc78e20) Python 3.13.1, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1 Locale: English/United States (en_US) Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
> 

Indeed, the fix is based on latest 1.0.1 :)

paullee0, 2025-02-22:
PR submitted - https://github.com/FreeCAD/FreeCAD/pull/19774

closed 2025-03-03 by yorikvanhavre

imchrm, 2025-05-17:
"**Bim.Door.Hole_Depth = Bim.Door..width**" 
This should be so because your design guarantees you can open the door. If there is a recess on the opposite wall, you will see it right away.

now (ver. 1.0.1)
Hole_Depth = Door.height (try to change the height of the door)