FreeCAD/FreeCAD #19564: Cannot determine the real Parent of a Link to a Body by selecting a Feature inside of the Link, in python.

leoheck, 2025-02-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting objects by clicking on 3D models, the last feature of a Body is selected, not the Body itself.
If I want to get the parent object, I can use `obj.getParentGeoFeatureGroup()`.
However, if the selected object is a Link to a Body, there is no direct way to get the Link object.

I can access the list of links from the parent object using `obj.getParentGeoFeatureGroup().InList()` to find the Link to the Body but only if I already know that I am looking for a Link.

However, if I don’t know whether I am selecting a Link or the real object, or if there is more than one Link, there is no way to identify the real Parent of the selected item.

Here is a sample file with these objects.
[parent_link_selection.FCStd.zip](https://github.com/user-attachments/files/18773493/parent_link_selection.FCStd.zip)

In other words. If a feature of a Body or a Link to a Body is selected, there is no way to get the selected Parent object. At least I could not find a way yet.

### Full version info

```shell
0.21, 1.0, 1.1-dev...
```

### Subproject(s) affected?

None

### Anything else?

The Subproject list (here on the Github's issue report) could be enriched with the PythonAPI, maybe.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-02-13:
This is more a forum question than a GitHub issue.
```python
sels = Gui.Selection.getSelectionEx("", 0)
sel = sels[0]
sel.Object  # The main object that was selected.
sel.SubElementNames  # Subelement names that describe the nesting. For example: ('Pocket.',)
```
If you need more help please consult the forum. I am closing this.

closed 2025-02-13 by Roy-043

leoheck, 2025-02-13:
> If you need more help please consult the forum. I am closing this.

That's true, you are right. Sorry to bother @Roy-043, I tried to find this answer without success for a long time. Thanks.

leoheck, 2025-02-13:
@Roy-043 your code works with my example above. But not with this next one.
The code you shared selects the Oldest (for lack of proper naming) Parent. I want to get the Newest Parent, the first immediate Parent of the selected element.

[immediate_parent_link_selection.FCStd.zip](https://github.com/user-attachments/files/18786162/immediate_parent_link_selection.FCStd.zip)

For instance, if a Body Feature is Selected (from a body or a link to a body), I want the access the Parent of it.

leoheck, 2025-02-13:
Asked the question here now.
https://forum.freecad.org/viewtopic.php?t=94767