FreeCAD/FreeCAD #19566: Moving Sketch geometry with Constraints hidden crashes or hangs up FC

steveklett, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting and moving sketch geometry with all the constraints hidden FC will hang or sometimes crash right away.
I was unable to reproduce with a new file, but can reproduce with the attached file.

Reproduction Steps:

1. Open the attached file
2. Hide part "A1" (Part001)
3. Edit Sketch "Base" (Sketch008)
4. Click the eyeball icon on the Constraints pane to show and then hide all constraints
5. Click Fit All to frame all geometry
6. Select an edge on the left geometry and attempt to move it
7. FC will hang or crash

If you do the same without hiding the constraints, it works fine.

[bookcase - RnD on corner vert partition - take 4 - Using Parts.FCStd.tar.gz](https://github.com/user-attachments/files/18776699/bookcase.-.RnD.on.corner.vert.partition.-.take.4.-.Using.Parts.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-14:
Cannot confirm. No lag, no error, no crash.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.7
  * CurvedShapes 1.0.13
  * Curves 0.6.55
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.5
  * Silk 0.1.5

```

luzpaz, 2025-03-16:
@steveklett can you reproduce on 1.1dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds