FreeCAD/FreeCAD #19570: Pressing F4 changes the layout of FreeCAD irreversibly and unpredictably!

raindropsfromsky, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If we press F4 repeadly, the  layout of FreeCAD is changed irreversibly and unpredictably!

For example, suppose that at the beginning, we have set Tree and Property panels on the lft, and the Tasks panel on the right.
Now, if we press F4 twice, the FC interface should look exactly as before.

But in reality, the FC interface changes drastically as  shown below:

[image]

 

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.8.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.1.0
  * SearchBar 1.3.3
  * sheetmetal 0.7.11
  * Silk 0.1.5
  * symbols_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-13:
I think it has already been reported here: https://github.com/FreeCAD/FreeCAD/issues/18049#issuecomment-2645562715

raindropsfromsky, 2025-02-13:
> I think it has already been reported here: [#18049 (comment)](https://github.com/FreeCAD/FreeCAD/issues/18049#issuecomment-2645562715)

Not at all!

He had a problem with F3/F4, as he ended up hitting a nearby button. Ultimately, he discovered that the **Tools > Customize** window allows him to set his own shortcuts, and he cleared off both shortcuts.

He never mentioned that the F4 button does not TOGGLE the state of the panels.
(By "toggle", I mean that pressing F4 twice should restore the layout exactly as before.)

semhustej, 2025-02-13:
Sorry, that's my bad. Did not read the OP carefully.

APEbbers, 2025-02-18:
Hi @raindropsfromsky,

Can you try if the issue is solved with either v1.7.3.5 or the develop branch of the Ribbon UI?

kind regards

raindropsfromsky, 2025-02-19:
> Can you try if the issue is solved with either v1.7.3.5 or the develop branch of the Ribbon UI?

No. There is no change.
Using 1.8.x

TomPcz, 2025-02-19:
The issue (panels changing position) is present even in a clean FreeCAD installation (no Ribbon or other addons).

TomPcz, 2025-02-19:
> Is this a Windows issue

No, the same issue happens in Linux, both in 1.0.0 and the latest weekly build.

```
OS: openSUSE Tumbleweed (KDE/plasma6/xcb)
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Czech/Czech Republic (cs_CZ)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 107.48/105.787
```

constint, 2025-04-13:
Also hit this issue and had to drag elements of the ui around to fix it. It's very jarring please fix if you can 🙏 
```
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * fasteners 0.5.38
```

luzpaz, 2025-05-07:
@raindropsfromsky @TomPcz @constint 
Can this bug be reproduced in Safe Mode[^1] ?

[^1]: https://wiki.freecad.org/Std_RestartInSafeMode

TomPcz, 2025-05-13:
> Can this bug be reproduced in Safe Mode

Yes.

**Steps to reproduce:**
* Delete all FreeCAD data (`~/.local/share/FreeCAD`, `~/.config/FreeCAD`) to start with a clean installation
* Download AppImage (here, it was `FreeCAD_weekly-2025.05.06-Linux-aarch64-py311.AppImage`, but the behaviour is the same with 1.0)
* Run the AppImage (with or without `--safe-mode`, makes no difference)
* Create a new file
* Press F4 (or click `Toggle overlay for all` in the menu) multiple times

**Expected behaviour:**
F4 is labelled as "Toggle overlay for all". As a new FreeCAD user, I am not sure what it is supposed to do. However, since it is "toggle", I would expect it to toggle between two defined states.

**Actual behaviour:**
Each time F4 is pressed, the layout changes randomly, it never goes back to the initial state.

https://github.com/user-attachments/assets/6fe429f9-ce6a-4885-9975-dadd5d44c28d

```
OS: openSUSE Tumbleweed (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.41585 (Git) AppImage
Build date: 2025/05/06 03:23:37
Build type: Release
Branch: main
Hash: 261ce705a93e98826389f20c4a99238757dfaab3
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/105.787/1
```

luzpaz, 2025-11-25:
Bumped the priority up for this since multiple users are reporting this.

kadet1090, 2025-11-25:
@luzpaz should we just disable all shortcuts for overlay? That seems like reasonable thing to do, not sure if removing the commands is a good thing but hitting it be accident should not be possible.