FreeCAD/FreeCAD #19571: Placement not fully updated when programmatically setting the AttachmentOffset to the origin's vertex

galou, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The following script fails to set the `Placement` of `LCS` correctly:
```python
doc = App.activeDocument()
part = doc.addObject('App::Part', 'Part')
lcs_obj = doc.addObject('Part::LocalCoordinateSystem', 'LCS')
lcs_obj.Visibility = True
part.addObject(lcs_obj)

attach_to_origin_vertex = True
if attach_to_origin_vertex:
    # This does not work:
    lcs_obj.AttachmentSupport = [
            (part.Origin, f'{part.Origin.OriginFeatures[-1].Name}.')]  # The origin point.
    lcs_obj.MapMode = 'Translate'
else:
    # This works:
    lcs_obj.AttachmentSupport = [
            (part.Origin, f'{part.Origin.OriginFeatures[3].Name}.')]  # The XY plane.
    lcs_obj.MapMode = 'FlatFace'

lcs_obj.AttachmentOffset = App.Placement(App.Vector(100, 0, 0), App.Rotation(App.Vector(0, 0, 1), 90))
doc.recompute()
```

Two intersting things:
- Attaching to the XY plane in the script works as expected (i..e. setting `attach_to_origin_vertex = False` in the script)
- Attaching to something else and then attaching to the origin's vertex works in the UI as expected (i.e. doing via UI was was supposed to be done via the script)

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (sway-nvidia/xcb)
Architecture: x86_64
Version: 1.1.0dev.40127 (Git)
Build type: Release
Branch: main
Hash: 20ac96f4c8a61eb5c8f3c1f78faa8c88a569165c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96.0344/154.858
Installed mods: 
  * freecad.startup
  * freecad.cross 1.0.0
  * Assembly4 0.50.17
  * Curves 0.6.58
  * boltsfc 2022.11.5
  * freecad.gears 1.3.0
  * CurvedShapes 1.0.13
  * Silk 0.1.5
  * fasteners 0.5.37
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
  * Defeaturing 1.2.3
  * RobotDescriptor
  * FreeCAD-AI-Toolbar
  * MeshRemodel 1.10.36
  * MnesarcoUtils 0.2.15
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-14:
@PaddleStroke FYI

galou, 2025-04-23:
I discovered a situation that is maybe related:
- create a Body
- attach an LCS to the body's origin
- attach a Sketch to the XZ plane of LCS
- the sketch has the wrong placement

```python
from FreeCAD import Vector, Placement, Rotation

def make():
    doc = App.newDocument()

    Body = doc.addObject('PartDesign::Body', 'Body')

    LCS = doc.addObject('Part::LocalCoordinateSystem', 'LCS')
    Body.addObject(LCS)
    LCS.AttachmentOffset = Placement(Vector(1.0, 0.0, 0.0), Rotation (0.0, 0.0, 0.2588190451025207, 0.9659258262890683))
    LCS.AttachmentSupport = [(doc.Origin, ('XY_Plane.'))]
    LCS.MapMode = 'FlatFace'
    LCS.ViewObject.Visibility = True
    Sketch = doc.addObject('Sketcher::SketchObject', 'Sketch')
    Body.addObject(Sketch)
    Sketch.AttachmentSupport = [(doc.XZ_Plane001, (''))]
    Sketch.MapMode = 'FlatFace'
    
    doc.recompute()

make() 
```

You get
```python
>>> doc.Sketch.Placement
Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,90)]
```
which corresponds to `doc.XZ_Plane001.Placement` whereas it should be `doc.LCS.Placement * doc.XZ_Plane001.Placement`.

galou, 2025-04-23:
A work-around previous issue is to add a plane attached to `XZ_Plane001` and attach the sketch to this plane.