FreeCAD/FreeCAD #19576: Create a 2D nesting tool

yorikvanhavre, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Nesting is the process of arranging different shapes inside one bigger container shape. This is often used in CNC or laser cutting operations, to try to figure out how to cut several shapes out of a piece of material, while wasting as little as possible of the material.

FreeCAD currently [has a nesting tool](https://wiki.freecad.org/Arch_Nest), but it is very inefficient and fails often.

This project should:

* Explore and assess the current tool, where it works and where it fails
* Investigate existing, open-source nesting solutions
* Find out if any is usable in FreeCAD, both technically (code-wise) and legally (compatible license)
* Research possible algorithms, or ways to better the current algorithm
* Decide either to use an external solution, or build or refactor our own algorithm
* Adapt the current tool to use it

### Difficulty

Medium/Hard

### Workbenches/Technologies

CAM/BIM, OpenCasCade, C++/Python

### Project size

350h

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

j-wiedemann, 2025-02-13:
This tool work great (the v2 branch) :
https://github.com/DanielLiamAnderson/Packaide/tree/v2
its gpl v3
I have use it in a macro vith FC 0.21 with good results. I will share it if any interest.

tarman3, 2025-02-13:
> This tool work great (the v2 branch) : https://github.com/DanielLiamAnderson/Packaide/tree/v2 its gpl v3 I have use it in a macro with FC 0.21 with good results. I will share it if any interest.

Please create topic in forum about this solution

pitch-circle, 2025-02-14:
There is also deepnest.io, which says it is open source.

Another one is svgnest, from which deepnest was developed.

A third, Rhino specific one is BlazingNest.

CADialog, 2025-02-15:
I think this is a functionality that should be addressed; additional points to those already listed could also be taken into account:

- creating a small report on material usage
- creating separate cutting plans
- consideration of rubble parameters such as width, distances and safety distances, permitted rotation angle of the parts, direction of progression (grain in wood).
- determination of the geometry to be occupied (types of geometry)
- management of the quantity of parts to be produced and raw materials

Yes, of course others can already do that, maybe you can fall back on it somehow or create integrations.

Greeting,
Frank

jffmichi, 2025-02-15:
If I might add another point to the wishlist:

- allow arbitrary shaped stock

Use case:
Cut a few parts out of a rectangular sheet of wood/aluminium/etc. for a project. In the next project, roughly sketch the remaining stock and nest parts into it instead of starting with a fresh sheet.

j-wiedemann, 2025-02-17:
You can test the macro that use Packaide here : https://github.com/j-wiedemann/freecad_nesting_packaide

Dexus, 2025-03-13:
https://github.com/deepnest-next would be my first port of call rather than deepnest.io which has been discontinued for 7 years.

Valera005, 2025-03-16:
Hello, I am interested in that project idead for GSOC. Is there anybody else discovering and writing project proposal for that project idea?

sliptonic, 2026-10-15:
Good work is being done on a nesting workbench.  There's a discussion that details many nesting features.  I see no reason to keep this issue open so I'm closing it.

closed 2026-10-15 by sliptonic