FreeCAD/FreeCAD #19578: BIM: Arch_Axis corrupted when 'Angles' set to 90 or 270

semhustej, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1.Open new BIM file
2.Create one axis
3.Select axis and go Data tab
4.Change `Angles` property to 90 (or 270)
5.Axis rotates but is infinitely long and is not displayed properly when view is rotated around

I am able to reproduce in FC 1.0.0

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

BIM

### Anything else?

The issue was discovered by another user in this forum thread: https://forum.freecad.org/viewtopic.php?t=94742

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-02-26:
This is a basic trig problem. The length of lines with a non-zero angle is calculated to make sure their ends have the same Y=Length coordinate (in the LCS) as zero angle lines. If the angle is 90 or 270 degrees the axis line does not intersect this Y=Length horizontal boundary.

The problem is obviously solvable, but the question is on what should the length of these axes be based?

semhustej, 2025-02-26:
I am not sure about this, but wouldn't it make sense to limit the `Angles` value to:
-90 < `Angles` < 90
 
If I understand it correctly, you would use Arch_AxisSystem to create standard ABC - 123 grid.

If the `Angles` property is meant primarily to cover cases like this, than there is no need to have angles over 90

[image]

It seems to me it would be best not to spend much development time on this.

Note: I have very little experience with axis in FreeCAD. I mostly used only standard perpendicular grid systems in my practice. I would not have the need for this functionality in my practice, because you can rotate the whole axis using `Angle` property.

semhustej, 2025-06-27:
Tested with the latest weekly and it is somewhat more acceptable than the original behavior. As this is a corner case in my opinion, it is good enough. Thank you.
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1

```