FreeCAD/FreeCAD #19579: BIM: Moving or copying Arch_axis  shows error

ErtugrulEge, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Steps to reproduce:
1.Open new BIM file
2.Create one axis
3.Move or copy axis

`19:45:50  <QuickMeasure> QuickMeasure.cpp(97): 0000019B47410170 : Standard_TypeMismatch: TopoDS::Edge`

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/127.628
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-13:
I can confirm the issue.
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library

```

yorikvanhavre, 2025-02-18:
Confirmed. This is not a BIM error, though, but in the quick measurement system.

I don't know why it yields an error here. because it's a pretty standard Part::FeaturePython object... 
Don't know who could help us here, maybe @PaddleStroke or @Syres916 ?

Roy-043, 2025-06-28:
AFAICT this has meanwhile been fixed. @ErtugrulEge can you check again? Thanks.

```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

ErtugrulEge, 2025-06-28:
I am closing the issue.Tnak you.

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38

closed 2025-06-28 by ErtugrulEge