FreeCAD/FreeCAD #19580: Door creation tool creates a Window instead

Slals, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Earlier I could create door easily, no problem. But for some reason it's now impossible to create doors, it only creates windows. Even in a blank BIM project with a fresh new wall.

I tried to reinstall Freecad but no luck...

I had an error when clicking on door tool, but I don't have it anymore. I can retrieve it from logs if there is any.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.65
```

### Subproject(s) affected?

BIM

### Anything else?

I clicked on "Door" tool, as you can see in the sidebar, there is settings for creating a Window...

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-02-13:
https://wiki.freecad.org/BIM_Door

semhustej, 2025-02-13:
The door tool behaves the same as the window tool. Only few of the settings are saved independently.

There should be the same presets available to both door and window tools:

[image]

This issue seems like a duplicate of https://github.com/FreeCAD/FreeCAD/issues/19101 to me.

Slals, 2025-02-13:
Okay good catch! I know understand the idea to merge both objects in the UI but it can be confusing!

closed 2025-02-13 by Slals

semhustej, 2025-02-14:
> Okay good catch! I now understand the idea to merge both objects in the UI but it can be confusing!

Yes, it is confusing. There is a plan to refactor the Window and Door tool. I think in the future it will be more separated to not confuse users.

yorikvanhavre, 2025-02-14:
Yes. the idea is to split the list of presets, and only show Door presets to the Door tool, and Window presets to the Window tool, and separate all the other settings (H1, H2, etc...)