FreeCAD/FreeCAD #19581: Surface - Fill boundary curves not honor SplitCurve edge if it's part of BSpline

Shkolik, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Have a problem with using " Fill boundary curves" in some cases. It works fine if all edges came directly from sketches, it works if one boundary edge is SplitCurve (from Curves WB) based on simple geometry like line or Arc, but doesn't when SplitCurve based on BSpline. Preview of selected edges show correct names and 3D view shows correct boundaries, but surface itself is obviously wrong.

[image]

Weirdly enough that "Filling" tool works fine with same boundary edges.

[image]

The problem is pretty old - was able to reproduce it on year old weekly build, so not sure when it started, nut it's pretty annoying.
And in some cases there is no workaround - one of the boundaries more likely will be edge of generated geometry and there is no way to trace it with 2 separate BSplines so it will match precisely, so I have to split it.

Example file (remove .zip extension):

[surface with splitcurve.FCStd.zip](https://github.com/user-attachments/files/18788065/surface.with.splitcurve.FCStd.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * Assembly3 0.12.2
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * fasteners 0.5.35
  * Foamcut 0.1.6
  * freecad.gears 1.3.0
  * Marz 0.1.15
  * OpenTheme 2024.9.1
  * PieMenu 1.11.2
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Shkolik, 2025-02-14:
Actually looks like it could be fixed in CurvesWB by creating copy of BSpline before splitting, but still, looks like a bug in Surface WB since it's inconsistent between 2 similar tools - GeomFillSurface and Filling.

maxwxyz, 2025-02-14:
@tomate44 is this an issue with CurvesWB or SurfaceWB?

tomate44, 2025-02-14:
It is a SurfaceWB bug.
At these line : 
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Surface/App/FeatureGeomFillSurface.cpp#L281
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Surface/App/FeatureGeomFillSurface.cpp#L331
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Surface/App/FeatureGeomFillSurface.cpp#L342
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Surface/App/FeatureGeomFillSurface.cpp#L355
you can see that the full untrimmed curve is used.
I think they need to be trimmed with something like : 
```
BRepAdaptor_Curve adapt(edge);
bspline->Segment(adapt.FirstParameter(), adapt.LastParameter());
```

maxwxyz, 2025-02-14:
I'm closing this bug in the FreeCAD repo, thanks for checking!

tomate44, 2025-02-14:
> I'm closing this bug in the FreeCAD repo, thanks for checking!

Why ? As I said, I think this is a FreeCAD issue.

maxwxyz, 2025-02-14:
Sorry, misread!

Shkolik, 2025-02-16:
> I think they need to be trimmed with something like :
> 
> ```
> BRepAdaptor_Curve adapt(edge);
> bspline->Segment(adapt.FirstParameter(), adapt.LastParameter());
> ```

Actually you are right! And we don't even need BRepAdaptor_Curve to do trimming there - we already have u1 and u2, so just 

`bspline->Segment(u1, u2);`

will be enough and only if spline was "created" by DownCast. Only 2 lines of code fixes issue. 

Thank you for idea! I submitted PR that should do the trick.

closed 2025-02-24 by chennes