FreeCAD/FreeCAD #19585: Importing FreeCAD to macos python REPL gives `symbol not found in flat namespace` error.

ornamentist, 2025-02-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm trying to use the following Python script on an M4 mac (macos 15.3):

```
import sys

sys.path.append("/Applications/FreeCAD.app/Contents/Resources/lib")
import FreeCAD
```

And I'm getting this Python error:

```
Traceback (most recent call last):
  File "/Users/stu/tmp/fc/1.py", line 4, in <module>
    import FreeCAD
ImportError: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/FreeCAD.so, 0x0002): symbol not found in flat namespace '__Py_PackageContext'
```

Searching the forums this may be a result of the version of Python used to build the macos version?

### Full version info

```shell
OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 72/108
Installed mods: 
  * PieMenu 1.11.2
  * woodworking 0.21.2.33771.0.23.1
  * OpenTheme 2024.9.1
  * IconThemes
  * Curves 0.6.55
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ornamentist, 2025-02-13:
This was using Python 3.13.x

ornamentist, 2025-02-14:
Same issue with the 1.0.0 version.

Something must have changed between Python 3.11.x and Python 3.12.x as I see the error with Python versions >= 3.12:

```
uv run --python 3.13 1.py
Using CPython 3.13.0
Removed virtual environment at: .venv
Creating virtual environment at: .venv
Traceback (most recent call last):
  File "/Users/stu/tmp/fc/1.py", line 6, in <module>
    import FreeCAD as App
ImportError: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/FreeCAD.so, 0x0002): symbol not found in flat namespace '__Py_PackageContext'

~/t/fc❭ uv run --python 3.12 1.py
Using CPython 3.12.5
Removed virtual environment at: .venv
Creating virtual environment at: .venv
Traceback (most recent call last):
  File "/Users/stu/tmp/fc/1.py", line 6, in <module>
    import FreeCAD as App
ImportError: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/FreeCAD.so, 0x0002): symbol not found in flat namespace '__Py_PackageContext'

~/t/fc❭ uv run --python 3.11 1.py
Using CPython 3.11.11
Removed virtual environment at: .venv
Creating virtual environment at: .venv
finished
```

luzpaz, 2025-09-26:
@johnocfii can you reproduce on macOS ? TIA

JohnOCFII, 2025-09-27:
I'm not a regular Python user -- but the steps listed above did not give me an error with the most recent weekly build on my MacBook Pro w/Arm processor.

```
Python 3.11.13 | packaged by conda-forge | (main, Jun  4 2025, 14:54:05) [Clang 18.1.8 ] on darwin
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Start_Start',0)
>>> # Gui.runCommand('Std_QuickMeasure',1)
>>> # Gui.runCommand('Std_Workbench',5)
>>> # Gui.runCommand('Std_Workbench',5)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> ### Begin command Std_New
>>> App.newDocument()
>>> # App.setActiveDocument("Unnamed")
>>> # App.ActiveDocument=App.getDocument("Unnamed")
>>> # Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> # Gui.activeDocument().activeView().viewDefaultOrientation()
>>> ### End command Std_New
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> # Gui.runCommand('Std_DlgPreferences',0)
>>> import sys
>>> 
>>> sys.path.append("/Applications/Other\ Apps/FreeCAD-2025-09-24.app/Contents/Resources/lib")
>>> import FreeCAD
>>> 
```

```
OS: macOS 15.7
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * Telemetry 1.0.2
```

ornamentist, 2025-09-27:
I checked again with Python 3.13.7:

```
❭ uv run --python 3.13.7 python -c 'print("Hello from Python 3.13.7!")'
Hello from Python 3.13.7!
```

Then:

```
❭ uv run --python 3.13.7 python
Python 3.13.7 (main, Sep 18 2025, 22:52:34) [Clang 20.1.4 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
...
... sys.path.append("/Applications/FreeCAD.app/Contents/Resources/lib")
... import FreeCAD
...
Traceback (most recent call last):
  File "<python-input-0>", line 4, in <module>
    import FreeCAD
ImportError: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/FreeCAD.so, 0x0002): symbol not found in flat namespace '__Py_PackageContext'
```
FreeCAD details:

```
OS: macOS 26.0
Architecture: arm64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * woodworking 0.21.2.33771.0.23.1
  * OpenTheme 2025.5.20
  * IconThemes
  * Curves 0.6.63
```

Doing this might require strictly Python 3.11 and not a later version?

ornamentist, 2025-12-05:
Thanks for clarifying that. Can I ask if there’s a migration plan to newer versions of the bundled Python?