FreeCAD/FreeCAD #19586: Suggestion: The Help > Report issue menu should take the user to a disambiguation page first, to guide him to the correct GitHub page for filing his issue.

raindropsfromsky, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The **Help > Report issue** menu option takes us to GitHub's "Issues" page for FreeCAD.

However, most often the problem is in an addon (WB, theme, etc.), and not the main FreeCAD app.
So the user has to find the appropriate GitHub site first and then file his issue.

For example, for the last few days, I am filing this suggestion at-
https://github.com/APEbbers/FreeCAD-Ribbon/issues/
and NOT at-
https://github.com/FreeCAD/FreeCAD/issues

But most users do not know this distinction, and they end up filing issues at the FreeCAD's GitHub.
This creates a clutter, and increases the load of the FreeCAD developers.
From the user's perspective also, he does not get timely solution.

**Suggestion:**
The **Help > Report an issue** menu should first take the user to a wiki page that guides the user to the correct GitHub address.

This idea is similar to how Wikipedia provides a [dismabiguation page](https://en.wikipedia.org/wiki/Wikipedia:Disambiguation) when the user gets confused about where to look.

At the start of that page, there should be a brief explanation about how to find the correct component.
Then all the links should be provided for FreeCAD and its addons.

With this arrangement, the FreeCAD user will be able to get his resolution faster.
This will also help the FreeCAD developers by keeping the FreeCAD GitHub page free of misplaced complaints.

### Full version info

```shell
This isssue is about how to redirect the user to the correct GitHub page. It is not related to the FC version!
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

galou, 2025-02-14:
There is also another confusion:
- if you create a new Github issue, the "START HERE" option forwards you to the forum
- if you look at the [forum topic "How to submit issues here"](https://forum.freecad.org/viewtopic.php?t=91897) it is stated
"Issues management will still happen primarily on GitHub. [...] If you have a GitHub account, or if you don't mind creating one, please consider reporting your issue there instead of here"

So, it's not clear whether issues should be preferably posted on Github or on the forum.

What I personally do is that if I'm sure it's a bug (for example if I can reproduce it with a script and I'm convinced that it's linked to my particular setup), I post it on Github but if I need confirmation from other users that they can reproduce the issue, I post on the forum first.

ErtugrulEge, 2025-02-14:
I think firstly, users should be routed to forum, if they couldn't solve the problem then user can apply to github,because mostly  problems  are due to users, not knowing the program well. I always look for a solution on forum and then github.

raindropsfromsky, 2025-02-14:
Why not!
Then the disambiguation page should be provided in the forum.

My main idea was not to allow the user to post all issues in the FC's GitHub.