FreeCAD/FreeCAD #19587: Offset geometry on a sketch with two squares hangs FreeCAD on macos

ornamentist, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Load the attached FreedCAD file, then:

1. Open the sketch of two adjacent squares.
2. Select all elements.
3. Choose Offset Geometry.
4. Change from "arc" to "intersection".
5. Select an offset amount inside the left square.

At this point FreeCAD on macos hangs and needs to be force-quit.

With the same file, but the squares unconstrained the same operation crashes FreeCAD.

### Full version info

```shell
OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 72/108
Installed mods: 
  * woodworking 0.21.2.33771.0.23.1
  * OpenTheme 2024.9.1
  * IconThemes
  * Curves 0.6.55
```

### Subproject(s) affected?

Sketcher

### Anything else?

[test.FCStd.zip](https://github.com/user-attachments/files/18793366/test.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-14:
Same on Windows:

[image]
@PaddleStroke FYI

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.7
  * CurvedShapes 1.0.13
  * Curves 0.6.55
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.5
  * Silk 0.1.5

```

ornamentist, 2025-02-14:
Same problem occurs with the 1.0.0 release version.