FreeCAD/FreeCAD #19588: [!Attention!] Freecad deleted the project

big-vl, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The place on the disk ended, Freecad restored the project when opening and I, out of habit, pressed Ctrl-S
The project file is now empty, Freecad does not start after such a Crash writes 

```
[big-vl@ho ~]$ freecad
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Failed to compute left/right minimum bearings for "cursor.pcf"
Failed to compute left/right minimum bearings for "cursor.pcf"
Failed to compute left/right minimum bearings for "cursor.pcf"
Failed to compute left/right minimum bearings for "cursor.pcf"
/usr/include/c++/14.2.1/bits/stl_vector.h:1130: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

Logov is not appropriately, I managed to find a file cache, but not for today, it's good that I had a habit of making backups for previous dates, but 3 months of work at 0.

### Full version info

```shell
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-14:
[image]

big-vl, 2025-02-14:
> [image]

3 empty backup file?

maxwxyz, 2025-02-14:
You are also using a non-stable version.

big-vl, 2025-02-14:
> You are also using a non-stable version.

And I think that the effect will be the same, I used the version as soon as it came out. Show what has changed in this regard?

kaiwas, 2025-02-14:
In fact, the situation is sad.
And I support the opinion that the problem was closed in vain.

Such a combination of circumstances may be rare, but it would still not be superfluous if the program informed the user that the file cannot be written to the hard drive due to lack of space.

big-vl, 2025-02-14:
> In fact, the situation is sad. And I support the opinion that the problem was closed in vain.
> 
> Such a combination of circumstances may be rare, but it would still not be superfluous if the program informed the user that the file cannot be written to the hard drive due to lack of space.

At least I did not note that the file is empty, because it is easy to check and add a check after saving the document. That the file is not 0 byte, that would be enough!

tarman3, 2025-02-14:
Try to create a topic in the forum to attract more attention.
Perhaps one of the developers will notice this

oursland, 2025-02-15:
> Such a combination of circumstances may be rare, but it would still not be superfluous if the program informed the user that the file cannot be written to the hard drive due to lack of space.

It was not clear from the issue that you had run out of disk space and FreeCAD either crashed or ended successfully without notice.  If this is the case, I would request that you open an issue outlining how one can reproduce the issue.

FlachyJoe, 2025-02-15:
I confirm the issue. Steps to reproduce:

- Create a minimal 4kB ramdisk
```bash
sudo mkdir /tmp/ramdisk
sudo chmod 777 /tmp/ramdisk
sudo mount -t tmpfs -o size=4k myramdisk /tmp/ramdisk
```
- Fill the disk
```bash
dd if=/dev/zero of=/tmp/ramdisk/zeros
```
- Create a new doc, save it to the ramdisk and try to reload
```python
$ ./bin/FreeCAD -c
>>> App.newDocument("Test")
>>> App.ActiveDocument.saveAs("/tmp/ramdisk/Test.FCStd") # No error raised
>>> App.closeDocument("Test")
>>> App.openDocument("/tmp/ramdisk/Test.FCStd")
<Exception> Invalid project file: /tmp/ramdisk/Test.FCStd
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
    App.openDocument("/tmp/ramdisk/Test.FCStd")
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: Invalid project file: /tmp/ramdisk/Test.FCStd
>>> quit()
$ ls -l /tmp/ramdisk
total 4
-rw-rw-r-- 1 florian florian    0 15 févr. 12:14 Test.FCStd
-rw-rw-r-- 1 florian florian 4096 15 févr. 12:12 zeros
```

kaiwas, 2025-02-15:
@maxwxyz  
>...FreeCAD crashes...

If I understand correctly - the program does not crash, but silently cannot write the document. Probably a warning should pop up that the disk space has run out and the file cannot be written. (outside the console, because not everyone has the console open)

PR #19630 (2025-03-20): Core: raise exception when no space left on disk at document save
Fix #19588
closed 2025-03-20 by hyarion