FreeCAD/FreeCAD #19589: Software crashes frequently

sebaras, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I load a Freecad file, Main explorer software crashes frequently.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * CADExchanger
  * ExplodedAssembly
  * fasteners 0.5.35
  * Help 1.0.3
  * sheetmetal 0.7.1
```

### Subproject(s) affected?

None

### Anything else?

When I load a Freecad file, Main explorer software crashes frequently.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

sebaras, 2025-02-14:
[image]
This could also be in combination with older Std files that combines a Part-Design an d e tech-Draw content.
When I open older files, and try to change something in Part design, the posted failure is displayed.

maxwxyz, 2025-02-14:
Please add a file and steps how to reproduce