FreeCAD/FreeCAD #19595: BIM: Doors and windows are hidden when walls are Arch_Added

semhustej, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Feature request

Steps to reproduce:

- open file

[FC-join-wall-door.FCStd.zip](https://github.com/user-attachments/files/18802466/FC-join-wall-door.FCStd.zip)

- select Wall001 than Wall
- use `Arch_Add` tool
- Walls are joined into one container, but the window in the joined wall is hidden
- User has to manually unhide the Window/Door

See behavior in the video:

https://github.com/user-attachments/assets/6ed4191e-08f4-45b9-99f3-b98eb016cad2

[image]

The behavior can be reproduced in FC 1.0.0

**Desired behavior**
Windows/Doors should remain visible after the `Arch_Add` command is finished.
If the window is hidden before the `Arch_Add` command is started, it should remain in hidden state.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell ..0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-18:
This looks like a bug indeed.  I wonder if the windows should not be transferred to the host wall? This would make things more logical...

semhustej, 2025-02-18:
> This looks like a bug indeed. I wonder if the windows should not be transferred to the host wall? This would make things more logical...

I don't know what the basic concept of this hierarchy is, but it would make sense to me to leave the window as a child of the original wall ('child wall'). When walls are 'Arch_Added', user can still work with each of the wall object independently (move it, change properties). I think this is good situation.
Now if the window was moved to be child of the host wall and user would move the 'child wall', the window would not have the relation to this 'child wall' and would stay at it's place. 
I think the window should remain as a child of the 'child wall'. I think this is more intuitive for user to understand as the window is related to 'child wall' in real world.

PR #22260 (2025-07-07): BIM: fix visibility handling of objects hosted by additions
Fixes #19595.

PR to make the visibility of objects hosted by additions depend on the visibility of the parent. This solution is preferable to changing Hosts properties. The latter solution would be hard to undo in case additions are removed.
closed 2025-07-07 by yorikvanhavre