FreeCAD/FreeCAD #19598: Update license file in the repo to match https://wiki.freecad.org/License

furgo16, 2025-02-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The [license file in FreeCAD's GitHub repository](https://github.com/FreeCAD/FreeCAD/blob/main/LICENSE) states FreeCAD is licensed under the GNU Library General Public License, version 2.0. However,

- https://wiki.freecad.org/License => states that FreeCAD is LGPL 2.0 or superior
- https://www.gnu.org/licenses/old-licenses/lgpl-2.0.html => states that the GNU Library General Public License, version 2.0 has been superseded by the LGPL

To remove ambiguity and any mismatch between the code and what the wiki states, it might be worth updating the `LICENSE` file in the repo to be LGPL 2.0 or superior.

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I'm not an expert on licensing, I simply thought I'd highlight this, as I noticed it while updating the `license` field of the snap package.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2025-02-15:
Clause 13 of the LGPL permits us to state that you make use any later version, but we are not permitted to change the text of the license itself. The LGPL 2.0 is exactly what the LICENSE file contains, and no modifications are allowed.

furgo16, 2025-02-15:
I just thought I'd highlight it as I noticed it, but if it's not an issue, feel free to close, no worries!

closed 2025-02-16 by maxwxyz