FreeCAD/FreeCAD #19600: Core: result of getParent is inconsistent

Roy-043, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The result of `getParent` is inconsistent.

1. Open the attached [file](https://github.com/user-attachments/files/18806937/closeGroupWires_bug.FCStd.zip). Remove the .zip extension.
2. Paste the code below in the Python console.
3. Result: the Part is returned.
4. In the Tree view, drag the Group out of the Part and drop it on the document node.
5. Restore the original nesting: drop the Group on the Part.
6. Paste the code again.
7. Result: the Group is returned.

```python
obj = App.ActiveDocument.getObject("Wire006")
obj.getParent()
```

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/86.4394
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tiagomscardoso, 2025-03-07:
Hi, is there anyone already working on this bug? I managed to replicated and would be interested in solving it. Could someone assign the issue to me?

tiagomiguelcardoso, 2025-06-09:
I was not able to fix it sorry.
Could I be unassigned?

Roy-043, 2025-06-10:
Thanks for trying!

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/21feb19c8a6869d7ea555f0fec3e3df43de45023

getParent prefers the group directly containing the object, instead of depending on the InList order after moving it around.