FreeCAD/FreeCAD #19603: Tech Draw:View do not appear when exporting to svg, pdf by Macro

LKY0916, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I use macros to implement part view projection and PDF export, the view displayed in FreeCAD after executing the macro is complete, as shown in Figure 1; And the view is lost in the exported PDF. Excuse me, is there a problem with the macro I wrote? The macro code is below.

Figure 1：
[image]
Figure 2：

[image]
Macro code：
import FreeCAD as App
import FreeCADGui as Gui
import Part
import TechDrawGui

class Gear_Tech_draw:

    def __init__(self, step_path, template_path, output_pdf_path):
        self.step_path = step_path
        self.template_path = template_path
        self.output_pdf_path = output_pdf_path

    def draw(self):
        # Drawing scale
        page_scale = 2.0

        # New Initial Document
        doc = App.newDocument('TechDrawing')
        part = doc.addObject('Part::Feature', 'Part')

        # Import 3D model
        shape = Part.read(self.step_path)
        part.Shape = shape

        # Import drawing template
        page = doc.addObject('TechDraw::DrawPage', 'Page')
        page.Scale = page_scale
        template_path = self.template_path
        template = doc.addObject('TechDraw::DrawSVGTemplate', 'Template')
        template.Template = template_path
        page.Template = template

        # Add gear front view
        front_view = doc.addObject('TechDraw::DrawViewPart', 'FrontView')
        front_view.Source = part
        front_view.Direction = (0, 0, 1)
        page.addView(front_view)
        front_view.X = 100
        front_view.Y = 130
        front_view.LockPosition = True
        # View detail adjustment
        Gui.getDocument('TechDrawing').getObject('FrontView').HorizCenterLine = True
        Gui.getDocument('TechDrawing').getObject('FrontView').VertCenterLine = True
        Gui.getDocument('TechDrawing').getObject('FrontView').FaceColor = (255, 255, 255)

        # Open the interface
        Gui.Selection.addSelection('TechDrawing', 'Page')
        App.getDocument('TechDrawing').getObject('Page').ViewObject.doubleClicked()
        Gui.Selection.clearSelection()

        Gui.runCommand('TechDraw_ToggleFrame', 1)
        doc.recompute()

        # Export as PDF
        TechDrawGui.export([page], self.output_pdf_path)

step_path = "E:/marco/cylinder_gear2.step"
template_path = "D:/Program Files/FreeCAD 1.0/data/Mod/TechDraw/Templates/A4_Landscape_ISO5457_advanced.svg"
output_pdf_path = "E:/marco/CAD/output/test_page.pdf"
Gear_Tech_draw(step_path, template_path, output_pdf_path).draw()

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Chinese/China (zh_CN)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-15:
@WandererFan could you please take a look at the code if export is done correctly?

LKY0916, 2025-02-15:
> please share an example file. [@WandererFan](https://github.com/WandererFan) could you please take a look at the code if export is done correctly?
This is the example file.
[test.zip](https://github.com/user-attachments/files/18809494/test.zip)

WandererFan, 2025-02-16:
test.py does not wait for the separate thread to complete before starting the export, so there is no geometry to export.   

```
09:59:11  test.py calling recompute
09:59:11  FrontView is finding hidden lines
09:59:11  test.py calling export page
09:59:11  test.py back from exportPage  <<<<<<< page has been exported
09:59:11  FrontView has finished finding hidden lines  <<<<< geometry has been created
09:59:11  FrontView is extracting faces
09:59:11  FrontView has finished extracting faces
```

WandererFan, 2025-02-16:
You will need to do what TDTest/DrawViewPartTest.py does:

     ```
  doc.recompute()

        # wait for threads to complete before checking result
        loop = QtCore.QEventLoop()

        timer = QtCore.QTimer()
        timer.setSingleShot(True)
        timer.timeout.connect(loop.quit)

        timer.start(2000)   #2 second delay
        loop.exec_()
        # carry on with other processing

        # Export as PDF
        TechDrawGui.export([page], self.output_pdf_path)
```

LKY0916, 2025-02-17:
> You will need to do what TDTest/DrawViewPartTest.py does:您需要执行 TDTest/DrawViewPartTest.py 执行的作：
> 
> ```
>  ```
>    doc.recompute()
> 
>     # wait for threads to complete before checking result
>     loop = QtCore.QEventLoop()
> 
>     timer = QtCore.QTimer()
>     timer.setSingleShot(True)
>     timer.timeout.connect(loop.quit)
> 
>     timer.start(2000)   #2 second delay
>     loop.exec_()
>     # carry on with other processing
> 
>     # Export as PDF
>     TechDrawGui.export([page], self.output_pdf_path)
> ```

It is indeed a thread issue. Thank you very much for answering my question. I am extremely grateful.

closed 2025-02-22 by WandererFan