FreeCAD/FreeCAD #19604: PartDesign: Hole - Custom clearance without having the thread modeled.

leoheck, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using PartDesign Hole, I want to use the Custom Clearance without having the Thread Modeled.

Right now, this option appear only when the Model Thread is checked.

I want to have the Custom Clearance for the standard M3 hole. Is this possible? Why? Why not?
[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-15:
@leoheck it is already possible, please use the latest weekly:

[image]

closed 2025-02-15 by maxwxyz

leoheck, 2025-02-15:
Thanks @maxwxyz . In your picture it looks like the "Drill angle" and the "include in depth" has some kind of glitch in the UI. Why the background is different? It looks buggy.

leoheck, 2025-02-15:
@maxwxyz can you please open the issue again, please?

[image]

There is no way to enter custom clearance. The option we have is the existing clearance it was just moved to a different place. It should be a custom clearance, since `standard`, `close` and `wide` means nothing. 
Also, what `standard`, `close`, `wide` means? There is hint or mention there.

For instance, I am makin holes to be 3d printed to use with screws/bolts. 
Then I have some M3 screws.
Then I set the Hole to be M3, but since I am 3d printing, I want the final hole to be a bit less than the space given for the hole. Generally, I use for M3 something like (3 mm * 0.92) to have a hole that is tight enough to fasten a M3 screw firmly.

In Freecad, I am using a custom hole to set the dimensions. But this does not show clearly the intention of using M3 as the screw diameter. A better approach would be to put the hole as M3, then a tolerance (in percentage maybe, the resulting diameter could be displayed as well) in another field.

Using this version right now.
``` 
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
```

leoheck, 2025-02-15:
This image widget is confusing. It gives us settings before the image, and also after it. Where the custom diameter is above the part to set the diameter.

The image should come at the beginning or the end but not in the middle of the settings.

maxwxyz, 2025-02-16:
@leoheck the clearance you try to set is the property for a custom thread clearance which makes sense if you model the thread. The presets for the non-threaded holes (Standard, Wide, Close) derive from the standard. It seems to me that you just want to make a hole with a custom diameter, which is possible by setting the standard to none.

leoheck, 2025-02-16:
I don't want to model the thread.
This is a bit useless and makes the design complex and heavy especially if it has a lot of screws.

These are the clearances measured when using the `Standard|Close|Wide` choices considering an M3 screw.

```
Size: M3
Clearance Standard: Radius 1.7 mm -> Diameter 3.4 mm
Clearance    Close: Radius 1.6 mm -> Diameter 3.2 mm	
Clearance     Wide: Radius 1.8 mm -> Diameter 3.6 mm
```

These tolerances make the screw pass through freely.

But I am requesting a "CUSTOM Clearance", where I can set a value that changes the standard diameter. So I can have something like:

```
Size: M3 (Using M3 here, the actual screw size, no "Custom" something)
Custom Clearance: 0.92 = M3 * 0.92 = 2.76 mm (diameter) <=== which less than the screw diameter
```

This way it is going to give me a hole that is smaller than the M3, so the M3 screw will be firmly attached to it while keeping the M3 set in the Size for readability and documentation instead of just reading "custom" word everywhere. In other words, the clearance is custom, not the screw size.

maxwxyz, 2025-02-16:
@FreeCAD/cad-advisory-group what is your opinion?

pierreporte, 2025-02-16:
[The new hole tool designed for cosmetic threads](https://www.figma.com/proto/nDkUKcH1YgriUSRG2EInYj/FreeCAD-Dialogs?node-id=247-3465&node-type=canvas&t=PXunFHu2h93heWMQ-1&scaling=min-zoom&content-scaling=fixed&starting-point-node-id=247%3A3465&show-proto-sidebar=1) (very similar to the new one already in FreeCAD) removed support for clearance holes. From experience, a dedicated input is not necessary for this. If we had to tweak the design to support them again, I would propose a checkbox near the diameter entry, that would display a set of entries below the diagram (screw standard, diameter, fit).

leoheck, 2025-02-16:
> I would propose a checkbox near the diameter entry

I understand you are talking about the structure of the widget, It can be improved, of course.

However, the discussion is to have a `Custom Clearance`, along with the existing clearances.
I want to have the hole shrunk a bit, the only options are predefined, and they are increasing the hole only.

pierreporte, 2025-02-16:
Basically the proposal only provides custom clearance.