FreeCAD/FreeCAD #19605: Add preference for viewport overlays (datums, planes, transform handle, axis cross)

maxwxyz, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As discussed in many issues and PRs, especially with the new core datums, it is not possible to adjust their default size.

There should be a single preference to control the UI elements which are overlayed in the 3D view, depending on the size of the frame of the 3D view window. So elements would behave as today, a constant size, independent from the zoom level.

The setting should control such elements, which should be displayed at a constant size, like
- Datum objects (Origin, LCS, Plane, Line)
- Transform dragger
- Axis cross

@FreeCAD/design-working-group FYI

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables-master 0.1.1
  * CfdOF 1.29.7
  * CurvedShapes 1.0.13
  * Curves 0.6.55
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.0
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.5
  * Silk 0.1.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-02-15:
It would be great to include https://wiki.freecad.org/Draft_WorkingPlaneProxy - they appear tiny on my system (without HiDPI monitor)

hyarion, 2025-02-15:
@quartzar I saw you mention that you have the same issue. Could you paste your full version info (from about).

I’m curious to see what freecad reads out as your screens dpi

quartzar, 2025-02-15:
Yep sure thing @hyarion :

```bash
OS: Ubuntu 24.04.2 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40176 (Git)
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/109.814
Installed mods: 
  * FreecadDiscordPresence 1.0.2
```

pierreporte, 2025-02-16:
Fully agree with the proposal. I would go further with the settings. Basically allow auto-expanding planes so that the old way is available (it’s a popular request), with an independent setting for base planes. This way, the new bigger planes displayed when creating a sketch could be dropped.

Roy-043, 2025-02-16:
> It would be great to include https://wiki.freecad.org/Draft_WorkingPlaneProxy on this list as well - they appear tiny on my system (without HiDPI monitor)

You can change the Arrow Size and Display Size of WP Proxies.

furgo16, 2025-02-17:
> You can change the Arrow Size and Display Size of WP Proxies.

Ah, I didn't know that, thanks!

Still, it is only changeable for individual WP Proxies through the View object in the Properties Editor. It would be great to have it as a global preference. Or to have them displayed at a fixed size independent of zoom, similarly to the other datum objects mentioned in this issue.

FEA-eng, 2025-05-31:
It would be good to implement it for the next major release because many new users might be confused by those tiny datums. Recently, I had to explain what this little blue rectangle actually is: https://forum.freecad.org/viewtopic.php?p=829809#p829797

hyarion, 2025-06-10:
> It would be good to implement it for the next major release

It's added to the 1.1 milestone and labeled as high priority so the current plan is to have this fixed before releasing 1.1.0.

kadet1090, 2025-08-12:
I have it working on some of my local branches. It indeed is important so I'll put PR for it before 1.1.