FreeCAD/FreeCAD #19609: Part Design Module - Filet probelm

jarwu78, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

It is not possible to thin-wall a solid that was extracted from, for example, a rectangle and then rounded its edges that are directly adjacent to the open plane.

### Full version info

```shell
It is not possible to thin-wall a solid that was extracted from, for example, a rectangle and then rounded its edges that are directly adjacent to the open plane.
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-15:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- Test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
- add full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

luzpaz, 2025-03-16:
Fillet bugs are very common. Did you search through the open fillet bugs to see if it's been reported ? 
https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue%20state%3Aopen%20fillet%20label%3A%22Topic%3A%20Fillets%22

Closing as there has been no response from IP

closed 2025-03-16 by luzpaz