FreeCAD/FreeCAD #19611: Astocad: Dragbar

Hoglet, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The dragbar functionality does not work in Linux (xfce)

### Full version info

```shell
Only way to move the window is with keyboard "modifier" (<Alt> in xfce)
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

This affects only the Astocad variant

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

depthoffocus, 2025-02-15:
> This affects only the Astocad variant

Then presumably the right place to raise this according to AstoCAD's own [FAQ](https://www.astocad.com/index.php/frequently-asked-questions/) is here?

https://github.com/AstoCAD/FreeCAD/issues

Hoglet, 2025-02-15:
Sorry

closed 2025-02-15 by Hoglet