FreeCAD/FreeCAD #19612: TechDraw_DraftView wrong scale

semhustej, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

TechDraw_DraftView shows all objects with wrong scale. This seems like a regression as I am not able to reproduce with FC 1.0.0. I think this was not present in weekly builds around mid to late january 2025.

Steps to reproduce:

- open the file 

[scale-testing.FCStd.zip](https://github.com/user-attachments/files/18809627/scale-testing.FCStd.zip)

- the TechDraw_DraftView shows the object in wrong scale
- I created TechDraw_View of the same object on the page to see the difference:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40239 (Git) Conda
Build type: Release
Branch: main
Hash: 3623f8baf50f924fd2a213495c72a5e316cf9341
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

TechDraw

### Anything else?

Related to this forum post: https://forum.freecad.org/viewtopic.php?t=94811

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-15:
see pr  #18757.

[image]

semhustej, 2025-02-15:
Thanks for looking into the issue. Changing the value of `LegacySvgScaling` in fact helps with TechDraw_DraftView being scaled properly.

In your comment for the PR, you mention: "The is no Gui update for this parameter, as it will only be changed in rare circumstances." But the issue I am reporting influences all existing and new TechDraw_DraftViews. That means that all users who utilize this tool would have to have `LegacySvgScaling` set to true all the time. I don't think this is a rare circumstance.

Steps to reproduce to see the influence:

- open FreeCAD in safe mode
- open sample file 'BIMexample'
- open page 'A4 Landscape'
- the scale of the drawing is wrong

[image]

The issue I am reporting appeared later than the https://github.com/FreeCAD/FreeCAD/pull/18757 was merged. I think it is not present longer than last three weeks.

ErtugrulEge, 2025-02-15:
LegacySvgScaling should be true as default 

[image]

WandererFan, 2025-02-16:
Looks like we need a finer control on the scaling.  legacy for Draft/Arch, new for other symbols.

closed 2025-03-03 by chennes