FreeCAD/FreeCAD #19615: BIM: Column object  Component->Axis Property doesn't work

ErtugrulEge, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Steps to reproduce:
1.Open file (File contains simple an axis and a column object)
2.Select column object and press  Component->Axis Property  and select the axis 
3.Column moves to other side

[column_axis_property.zip](https://github.com/user-attachments/files/18809986/column_axis_property.zip)

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-27:
@Roy-043's assessment is correct. You need to move the column yourself to where you want it on the axis. Then, it will be duplicated that way on subsequent axes. This allows for much more advanced positioning, as you can for ex. position an object not directly on the axis. Lots of combinations are possible, which wouldn't, if the object would be positioned automatically.

The thing is that's maybe unclear, is that the Arch Axis can have more than one axis. So there is some additional functionality there that is not so obvious when there is only one single axis. Maybe that's something that need to be better documented...

ErtugrulEge, 2025-03-01:
I think I get the logic. (Column move distance = Axis from origin distance)  so you have to create a column at the origin for aligning center on the axis.If you create column somewhere else it will move again the same distance as seen on the image

My opinion is, this is not a correct behaviour.When I choose axis, columns should move relative to axis.If I put next to axis, column should move with axis togehter.[Feature request]

In real building projects you have different size of columns and you can't center every column, please consider real case uses while programming.

Now,with the default axis property  behaviour, we can show a warning message to user about columns behaviour(E.g You have to move column where you want after selecting axis property)  when user press axis property

but this is not enough after moving column if you rotate the column, column moves to other location(second image)Similar behaviour

[image]

[image]