FreeCAD/FreeCAD #19618: BIM: Arch_Axis horizontal bubble text orientation is incorrect on 3D view

ErtugrulEge, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

In 3d view, Arch_Axis horizontal bubble text orientation is incorrect.If you make draftview on a TD page you will see the correct orientation but in 3D view they are incorrect

[image]

[image]

[horizontal_bubble_text_orientation.zip](https://github.com/user-attachments/files/18810039/horizontal_bubble_text_orientation.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
Is there a way to adjust orientation manually? Or is the intention to always show the labels upright in viewing direction? Are other labels and dimensions also affected?

ErtugrulEge, 2025-02-16:
I haven't seen a  property that changes it manually. This doesn't affect other labels or dimensions or any function but 

This is an architectural technical drawing convention.As far as I know,  horizontal axis  text position should be always like  TD page(second image).On TD page their position is correct. Please, look at the horizontal axis example in the link :
https://www.alamy.com/stock-photo-house-plan-blueprint-technical-drawing-part-of-architectural-project-143028920.html

maxwxyz, 2025-02-16:
@yorikvanhavre FYI

yorikvanhavre, 2025-02-18:
It's indeed a problem to be solved. There is a type of coin text that stays aligned to the camera, but it's really annoying to use because you can't scale it properly.

I see two possible solutions here:

1. We add a "Text Rotation" property that allows the user to fix the issue manually. Problem: It's annoying
2. We always rotate the text so it aligns vertically with the global Y axis. Problem: Users might look at their model from a different angle, and this might become worse than the current situation.

Maybe this:

Solution 1 + whenever the placement of the axis changes, the "Text Rotation" property is recalculated to match the global Y axis.

Roy-043, 2025-02-26:
Related: #9888.

ErtugrulEge, 2025-11-04:
Same issue continue

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251102 (Git shallow)
Build date: 2025/11/02 16:03:37
Build type: Release
Branch: main
Hash: 45ca23d074f39a9b586c8320f1e7cd20ca466fef
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```