FreeCAD/FreeCAD #19620: BIM: Arch_axis creates incorrect Shape2DView

ErtugrulEge, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Arch_axis creates incorrect Shape2DView , axis loses Draw Style and bubbles and numbers

Steps to reproduce:
1.Create axis objects
2.Create a folder
3.Select axis objects and move them into folder
4.Select folder and press BIM_Shape2DView 

[image]

also even Arch_AxisSystem creates an empty  BIM_Shape2DView 
Steps to reproduce:
1.Create axis objects
2.Select axis objects andf press Arch_AxisSystem
3.Select Arch_AxisSystem and press BIM_Shape2DView  will create an empty  BIM_Shape2DView  object

[axis_shape2dview_test.zip](https://github.com/user-attachments/files/18810201/axis_shape2dview_test.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
Did this work before 1.1?

ErtugrulEge, 2025-02-16:
Version 1.0 also has same pproblem

yorikvanhavre, 2025-02-18:
Indeed the arch axis was never made to be "Shape2Dview-able"... Basically, the shape2Dview only gather and projects Part shapes. However, the bubble and the number are not shapes, just coin3d nodes added to the 3D representation.

Since it's not possible to have texts projected anyway (in theory it is possible, if those texts are shapestrings, but that would be overkill and bring other many problems),  I don't think there is much that we can do here to solve the problem.

However, I indeed see the point here: One places axes in the model, and would like those axes to be reproduced on the plan view, without having to copy them.

To remedy that, we could think of some solution. Maybe a way to clone axes? So there would be one "Master" axis and linked copies?