FreeCAD/FreeCAD #19628: crash when trying to constrain symmetric

madbobmcjim, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm trying to follow this tutorial here:
https://www.youtube.com/watch?v=E14m5hf6Pvo

At 5:50 they suggest you select the select two points and the horizontal origin and press the symmetric constraint. When I do that FreeCAD crashes.

I'm running the current beta Flatpak version on Ubuntu 24.10. 

I've run FreeCAD through the CLI to reproduce and get an error message and it gives this error:

```
FreeCAD 1.1.0, Libs: 1.1.0devR39935 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
/usr/include/c++/13.2.0/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::vector<GCS::Constraint*>; _Alloc = std::allocator<std::vector<GCS::Constraint*> >; reference = std::vector<GCS::Constraint*>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
```

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.39935 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 49b304ae55)
Hash: 49b304ae558553f9acd5872ccd30673bbd60bd19
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
Could you add your file as an example and maybe record a macro up to that point where it fails?

madbobmcjim, 2025-02-16:
I can upload a file if you want, but it's literally 5 points in a sketch with a constraint added.
I've uploaded a video of me reproducing the issue to show what I mean.

adrianinsaval, 2025-02-16:
I maintain the flatpak, it's possible that this has been fixed in main, I'll push an update to flatpak beta

madbobmcjim, 2025-02-20:
I pulled the updated Flatpak and this issue went away, thanks :-)

closed 2025-02-20 by madbobmcjim