FreeCAD/FreeCAD #19629: BIM: change 'NativeIFC' to 'Native IFC' in preferences

semhustej, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

According to this topic https://forum.freecad.org/viewtopic.php?t=94403  'native IFC' will be the default naming convention instead of 'NativeIFC' on the Wiki. I think it would be appropriate to change the text also in Preferences:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40239 (Git) Conda
Build type: Release
Branch: main
Hash: 3623f8baf50f924fd2a213495c72a5e316cf9341
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
Installed mods: 
  * symbols_library
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-18:
Makes sense indeed. NativeIFC was mostly a working name (and a good hashtag :) ). Indeed the correct form should be Native IFC.

furgo16, 2025-02-18:
There you go, then => https://github.com/FreeCAD/FreeCAD/pull/19678

I hope this helps!

yorikvanhavre, 2025-02-18:
Super! Thanks @furgo16 !

closed 2025-02-18 by yorikvanhavre