FreeCAD/FreeCAD #19631: Part Extrude: Whether or not resulting part is solid is random or inconsistent

theosib, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have a partial observation here that I will add to as I make more careful observations.

When I make a sketch in Sketcher and extrude it in Part, whether I get a solid or just a vertical shell seems random. Sometimes I get a solid. Sometimes I get an outline shell. I can't find a pattern to it.

There is one important thing I'm not sure about. Is the "solid" state of the resulting extrusion different from the checkbox? Or is the value of the checkbox random? I don't know yet.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Logical/physical DPI: 72/127.5
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
Do you have an example file and steps how to reproduce. IIRC the checkbox is the same as the solid value in the property view. By default it will create solids on closed profiles and surfaces with open profiles.

theosib, 2025-02-17:
Someone else on Discord confirmed that he's observed that the "Solid" checkbox in the Extrude dialog box is randomly unchecked sometimes.