FreeCAD/FreeCAD #19632: Segmentation fault after run in console with Python code

VM4Dim, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I suppose after revision 40182 / commit 516595f / PR #18594

Reproduced with FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage
```
$ ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c 'print(2025)'
2025
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3fda0) [0x7f661e249da0]
#1  0x7f661d56bf07 in xercesc_3_2::XMLString::release(char16_t**, xercesc_3_2::MemoryManager*) from /tmp/.mount_FreeCAkehFlf/usr/bin/../lib/./libxerces-c-3.2.so+0x7
#2  /tmp/.mount_FreeCAkehFlf/usr/bin/../lib/libFreeCADBase.so(+0x164fd4) [0x7f661fd64fd4]
#3  /lib/x86_64-linux-gnu/libc.so.6(+0x42261) [0x7f661e24c261]
#4  /lib/x86_64-linux-gnu/libc.so.6(+0x4231a) [0x7f661e24c31a]
#5  /lib/x86_64-linux-gnu/libc.so.6(+0x29d6f) [0x7f661e233d6f]
#6  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f661e233e25]
#7  /tmp/.mount_FreeCAkehFlf/usr/bin/freecad(+0x6c7e) [0x55bcaf90fc7e]
```

### Full version info

```shell
-
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

VM4Dim, 2025-02-16:
Steps to reproduce for Linux.

Download FreeCAD_*.AppImage from: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
> wget https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage

Make it executable: 
> chmod +x FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage

Run:
> ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c "print('== Start =='); doc=FreeCAD.newDocument(); doc.saveAs('/tmp/Test.FCStd'); print('== End ==')"

Result:
```
== Start ==
== End ==
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3fda0) [0x7f75fe049da0]
#1  0x7f75fd36bf07 in xercesc_3_2::XMLString::release(char16_t**, xercesc_3_2::MemoryManager*) from /tmp/.mount_FreeCAPALbLm/usr/bin/../lib/./libxerces-c-3.2.so+0x7
#2  /tmp/.mount_FreeCAPALbLm/usr/bin/../lib/libFreeCADBase.so(+0x164fd4) [0x7f75ffb64fd4]
#3  /lib/x86_64-linux-gnu/libc.so.6(+0x42261) [0x7f75fe04c261]
#4  /lib/x86_64-linux-gnu/libc.so.6(+0x4231a) [0x7f75fe04c31a]
#5  /lib/x86_64-linux-gnu/libc.so.6(+0x29d6f) [0x7f75fe033d6f]
#6  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f75fe033e25]
#7  /tmp/.mount_FreeCAPALbLm/usr/bin/freecad(+0x6c7e) [0x55d9aaa27c7e]
```

File /tmp/Test.FCStd is created.

benj5378, 2025-02-16:
I can reproduce. The segfault only occurs when
```bash
$ freecad -c "print('any python command')"
```
and not when
```bash
$ freecad -c
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> "print('any python command')"
```

I'm not sure the `-c` option was ever meant to be followed by a Python command?

Flast, 2025-02-17:
I can't reproduce.
```
$ ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c 'print(4444)'
4444
```

```
$ ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> print(4444)
4444
>>> 
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> 
```

I also tried with gdb to trap any signal, but no signals caught.

```
$ gdb --args ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c 'print(4444)'
GNU gdb (Fedora Linux) 15.2-4.fc41
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage...

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.fedoraproject.org/>
Enable debuginfod for this session? (y or [n]) 
Debuginfod has been disabled.
To make this setting permanent, add 'set debuginfod enabled off' to .gdbinit.
(No debugging symbols found in ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage)
(gdb) r
Starting program: /home/pfn/.local/bin/FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c print\(4444\)
[Detaching after fork from child process 251748]
process 251745 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[Detaching after fork from child process 251755]
[Detaching after fork from child process 251757]
4444
[Inferior 1 (process 251745) exited normally]
Missing rpms, try: dnf --enablerepo='*debug*' install bash-debuginfo-5.2.32-1.fc41.x86_64 ncurses-libs-debuginfo-6.5-2.20240629.fc41.x86_64 glibc-debuginfo-2.40-21.fc41.x86_64
(gdb) quit
```

VM4Dim, 2025-02-17:
It reproduced with github actions:

```
name: Test-19632                                                                                                                                                              
                                                                                                                                                                              
on: workflow_dispatch                                                                                                                                                         
                                                                                                                                                                              
jobs:                                                                                                                                                                         
  Test_19632:                                                                                                                                                                 
    runs-on: ubuntu-22.04                                                                                                                                                     
                                                                                                                                                                              
    steps:                                                                                                                                                                    
    - name: Get AppImage                                                                                                                                                      
      run: |                                                                                                                                                                  
        wget -nv https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage                      
        sudo chmod +x FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage                                                                                           
                                                                                                                                                                              
    - name: Run AppImage                                                                                                                                                      
      run: |                                                                                                                                                                  
        ./FreeCAD_weekly-builds-40253-conda-Linux-x86_64-py311.AppImage -c "print('== Test-19632 ==')" || true                                                                
```

Result:
```
== Test-19632 ==
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f8095242520]
#1  0x7f809456bf07 in xercesc_3_2::XMLString::release(char16_t**, xercesc_3_2::MemoryManager*) from /tmp/.mount_FreeCAKMMjnJ/usr/bin/../lib/./libxerces-c-3.2.so+0x7
#2  /tmp/.mount_FreeCAKMMjnJ/usr/bin/../lib/libFreeCADBase.so(+0x164fd4) [0x7f8096f64fd4]
#3  /lib/x86_64-linux-gnu/libc.so.6(+0x45495) [0x7f8095245495]
#4  /lib/x86_64-linux-gnu/libc.so.6(on_exit+0) [0x7f8095245610]
#5  /lib/x86_64-linux-gnu/libc.so.6(+0x29d97) [0x7f8095229d97]
#6  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f8095229e40]
#7  /tmp/.mount_FreeCAKMMjnJ/usr/bin/freecad(+0x6c7e) [0x56268580cc7e]
```

Flast, 2025-02-17:
OK, I reproduced on Ubuntu 22.04 container, but not on host Fedora 41.

I'd like to debug this but, reverting #18594 might be better at this time.

Flast, 2025-02-17:
I don't know but @VM4Dim bisected and found: https://github.com/FreeCAD/FreeCAD/pull/18594#issuecomment-2655419305 .

VM4Dim, 2025-02-17:
> OK, I reproduced on Ubuntu 22.04 container

24.04 is same

luzpaz, 2025-03-16:
This is a regression, ITC ?

PR #20563 (2025-05-15): Base: Fixed segfault on destructing cached string
Xerces default memory manager is deleted before destructing static local variable and segfault.

This bug is introduced in #18594, reported at #19632.

closes: #19632
closed 2025-05-15 by chennes