FreeCAD/FreeCAD #19638: Part: "Negative solid" or "hole" option would save a ton of work

theosib, 2025-02-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If you're familiar with Tinker CAD, you'll know the way it does boolean cuts is by turning an object into a negative or "hole" and then grouping it with other objects.

Most of the time, the Cut boolean does the job just fine. But there have been many occasions where I had wished there was a way to set a "hold" or "negative" property on a solid so that unioning it with other solids would result implicitly in a cut. The circumstances mostly break down to these two cases:
- I want to group a bunch of objects together like a compound or union, but some of them are cuts, but I don't want to create extra levels of hierarchy.
- I have an existing hierarchy that contains an object I want to modify. If the modification is an addition, I can just drag more objects into the union. But if the modification is a cut, then I have to completely rip the entire hierarchy apart and build it all up again, just for a tiny modification. A negative property would save enormous amounts of work.

### Full version info

```shell
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Logical/physical DPI: 72/127.5
Installed mods: 
  * freecad.gears 1.3.0
  * Pyramids-and-Polyhedrons
  * fasteners 0.5.29
  * Curves 0.6.50
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
I assume this would need another operation then, otherwise users might be confused why some objects behave differently in a boolean addition.
@FreeCAD/cad-advisory-group FYI

pierreporte, 2025-02-16:
That would be like in Catia where you can make pockets out of nothing, and then assemble this “negative” body with a “positive” body. Basically it feels like a boolean operation but it’s not.

depthoffocus, 2025-02-17:
> I assume this would need another operation then, otherwise users might be confused why some objects behave differently in a boolean addition. @FreeCAD/cad-advisory-group FYI

I wonder, could the Hole tool not be modified to allow you to nominate a solid to cut with?

FC-FBXL5, 2025-02-18:
Any element can be created as a positive solid with a property "isNegative". A boolean tool "Assemble" could switch according to that property. If False the second element of a boolean OP is fused to the first, if True the second is cut from the first. The tool should also allow to change the result manually from Assembly to Union or Cut, this way one could also subtract a positive solid or add a negative one.

Assemble would respect the property while Union and Cut ignore it. And as long as a solid stands alone it is displayed as a positive solid. (I think nothing else does Catia)

A pocket is not made of nothing, it is simple a solid (with a property "isNegative" set to True) that becomes a hole when it is cut from another solid, like it is done in Part workbench.