FreeCAD/FreeCAD #19643: BIM: doors and windows doesn't follow walls that created with sketcher when sketch value changed

ErtugrulEge, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings to the Community!

Walls created with sketcher are losing their positions when sekcth value changed
Steps to reproduce:
1.Create a BIM file or use the attched file
2.Press BIM_Sketch
2.Create a simple rectangle with sketcher ie. 600 cm x 700 cm
3.Close sketcher
4.Select scketch and press Arch_Wall
5.Insert a door and window to  700 cm wall 
6.Select sketch and edit and change the wall size 600 cm to 400 cm
7.You will see door and window will lose their position in a wall

this problem makes parametric floor plan design very difficult.

[parametric_wall_test.zip](https://github.com/user-attachments/files/18813940/parametric_wall_test.zip)

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/127.628
```

### Subproject(s) affected?

BIM

### Anything else?

After updating the sketch, you can move doors and windows manually to new wall position but this  is very time consuming process.It is strange that it is difficult to make parametric simple floor plans while parametric complex objects can be easily made with sketcher.

As discussed on the forum @paullee0 has already some solution with add-on but I think this kind of feauture should be default.Ma

Forum topic : https://forum.freecad.org/viewtopic.php?t=94829 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

paullee0, 2025-02-16:
There is similar and/or associated discussion e.g. https://github.com/FreeCAD/FreeCAD/issues/14927 :)

To me, solution was already available but things like attributes' name, expectation of behaviour etc. to be polished.  Would like to make it available in stock FC BIM indeed.

closed 2025-02-16 by maxwxyz

paullee0, 2025-04-18:
@ErtugrulEge Check this PR / Forum Discussion:

- https://github.com/FreeCAD/FreeCAD/pull/20872
- https://forum.freecad.org/viewtopic.php?p=822493#p822493