FreeCAD/FreeCAD #19644: [BIM] BIM view. Should switch the active layer when changing the working plane.

kaiwas, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If we create two floors and switch between them using BIM view, we have to manually switch the active group. Otherwise, the objects are added to the view tree on the wrong floor.

Reproduction method.
1. Create two floors at different levels.
2. Double-click on the desired floor. In this case, the working plane moves to the desired floor. But the floor does not become active. If another floor was previously set, the objects will be set at the desired height, but in the view tree they will belong to the wrong floor.

For the correct algorithm of actions, we must always do two actions. Move the working plane to the desired floor and manually activate this floor. It would be better if the second action were done automatically.

Video demonstration.

https://github.com/user-attachments/assets/c171efb2-c0de-4006-96d8-5baf63d83f54

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * PitchedRoof
  * Road 2025.2.6
  * addFC 1.3.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.11
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-02-18:
BuildingParts (Floors and Buildings) have an "activate" property somewhere I don't recall (in their view properties?). When that property is turned on, when double-clicking the object in the tree, it will become the active container.

Could you check:

1. If that works
2. If the group drop-down is updated

That would help narrowing down the problem...

kaiwas, 2025-02-18:
I tested the work again. Indeed, if I set "Activate with double click" - then I get the desired result. That is, objects are added to the desired floor, even if the layer is not activated in the panel.

This is a partial solution to the problem, but there is still a little confusion. The floor is active, but it is not displayed in the active layers.

https://github.com/user-attachments/assets/3bbd38f8-b324-4ed5-b431-c9a953754e47

I would also like to see this setting in the global settings or available by default. Because it is really convenient and allows you to add objects to the desired level without dragging them into the tree view.

kaiwas, 2025-02-18:
Found a small peculiarity. At the active level, objects are added where needed. But groups are dropped. Probably this is a topic for another issue.

https://github.com/user-attachments/assets/5e20dfdf-7bb2-4168-8933-fc60d3c1bbb9