FreeCAD/FreeCAD #19646: assembly Workbench doesn't work in Linux Mint 22 Cinnamon 6.2.9, Linux Kernel 6.8.0-51 generic

KUKartist, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Assembly is highlighted green, not yellow. I can insert the parts, then save them, but not move them individually. Joint are activated.  Create joint leads to arbitrary results, parts cannot be moved, Freecad crashes.

[image]

#OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * OpenTheme 2024.9.1

### Full version info

```shell
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Please add an example file and steps how to reproduce

drwho495, 2025-02-23:
Please also try 1.1dev

luzpaz, 2025-03-16:
> Please also try 1.1dev

@KUKartist 👆