FreeCAD/FreeCAD #19647: TechDraw: Broken View not collapsing

josef-dijon, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Inserting broken view in TechDraw creates the view, but doesn't collapse it correctly in the drawing.

I followed the steps in the FreeCAD docs to create the broken view, and everything makes sense there. I've attached a screenshot below. Notice that the length dimension is also incorrect, I assume because it measures the total distance in the view, plus whatever distance the break represents.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.31
  * fcgear
  * freecad.gears 1.3.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

TechDraw

### Anything else?

Screenshot of my particular scenario:
[image]

Hunting through the existing issues I can't see anything like this reported already, although I did see this one: https://github.com/FreeCAD/FreeCAD/issues/15876 which is exploded view related, but has a similar feel where drawing elements are not offset correctly or something.

Attaching my part file too:

[ballscrew.zip](https://github.com/user-attachments/files/18814475/ballscrew.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

josef-dijon, 2025-02-16:
My first suspicion is in this function.

https://github.com/FreeCAD/FreeCAD/blob/de9bd5e1ce5c4b69a9e4b0e9bbb0ae63c9565633/src/Mod/TechDraw/App/DrawBrokenView.cpp#L252

If someone can point me at a document for compiling/debugging under Windows, I'll be happy to take a look.

WandererFan, 2025-02-16:
I'm not a windows guy, but this looks promising: https://github.com/FreeCAD/DevelopersHandbook/blob/main/gettingstarted/VSCode.md

We always need more hands on the keyboard!

josef-dijon, 2025-02-17:
I've created an MRE and attached it here.

It appears that the broken view works correctly up until the subtractive helix is applied.
 
[minimum_example.zip](https://github.com/user-attachments/files/18818030/minimum_example.zip)

WandererFan, 2025-02-18:
The check in moveThisShape() is for the edge of the broken piece being less than Precision::Confusion() (1e-7) from a break point.  In this case the piece is exactly 1e-7 away, so moveThisShape() wrongly says do not move this piece.

josef-dijon, 2025-02-18:
Nice! Thanks @WandererFan, your work is greatly appreciated :)

I only just got the FreeCAD libpack compiling yesterday, so this was always going to take me longer to get to.

WandererFan, 2025-02-19:
I have a long todo list now that you have your build environment set up. ;)

closed 2025-03-03 by chennes