FreeCAD/FreeCAD #19648: OFFSET- selecting multiple lines & vertices results in one vertice/line not offsetting

JDMils, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

My forum post here: [https://forum.freecad.org/viewtopic.php?p=811190#p811190](url)
So when creating an offset of a square wave shape, the first line in the shape (the one on the left) is NOT offset and is thus hidden by the original line. This line is actually pointing in the wrong direction compared to the original line, however still not offset to the original.

From what others are posting, it seems this is an existing problem which I cannot find the issue request for the fix.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * Curves 0.6.54
  * fasteners 0.5.36
  * Gridfinity 0.8.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-22:
Please add an example file and steps how to reproduce

JDMils, 2025-02-23:
Not sure why but it seems the link to the forum post I originally posted above is not working. Please check out my forum post here: [](https://forum.freecad.org/viewtopic.php?p=811190#p811190)

My example model is attached to the forum post- I hope this is OK with you?

The example was simply created by creating a shape consisting of a square wave and offset it using the Draft | Offset feature. You will notice that the "offsetted" horizontal line to the left of the screenshot is "missing" however it is actually behind the original line and now instead of pointing away from the model, it is now in the direction going into the model. So no offset for this one line.

FEA-eng, 2025-09-29:
From what I see, the horizontal line is there but it's short and inverted for some reason:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

JDMils, 2026-05-25:
I think this issue is fixed now? I can now produce offsets of the same shape as in the original post.
`OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260505 (Git shallow)
Build date: 2026/05/05 22:34:03
Build type: Release
Branch: (HEAD detached at 08cacbbc2)
Hash: 08cacbbc27421e769be4ec423f9815a01e51e533
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Australia (en_AU) [ OS: English/Australia (en_AU) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
Installed mods:
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Assembly4 0.60.6
  * Curves workbench 0.6.71
  * Fasteners Workbench 0.5.44
  * Gridfinity Workbench 0.12.4
  * Lattice2 Workbench 1.1.0
  * Manipulator Workbench 1.6.4
`
[image]