FreeCAD/FreeCAD #19650: Draft: Editing objects does not tell the user what to do

maxwxyz, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When editing a Draft object - to adjust individual points - there is no task dialog or information on what the user can do or how to edit it. According to @yorikvanhavre this is a regression.

When editing a line, this is the task dialog, it just says close:

[image]

### Full version info

```shell
OS: macOS 13.7.3
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/72
Installed mods: 
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * CfdOF 1.29.8
  * Defeaturing 1.2.3
  * Manipulator 1.5.7
  * fasteners 0.5.37
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.11
  * Curves 0.6.58
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
@Roy-043 FYI

Roy-043, 2025-02-16:
This is not a regression or a bug, but the result of a design decision during the V0.19 dev cycle. At that time Draft Edit was overhauled, and the task panel modified. I am not sure what the reasons were, I was not involved with FreeCAD programming at that time. I agree that the current situation is confusing. Note that this empty task panel is also used when modifier commands are invoked without a prior selection.

yorikvanhavre, 2025-02-18:
Basically the Draft Edit tool allows to do one thing: Move points.

Maybe this:

- While no point is selected: A text that says "Please click an editable point"
- While a point is selected: Display the Draft point coordinates UI

Roy-043, 2025-02-18:
Draft Edit can do several other things as well. Some options depend on the selected object. The decision to use context menus makes sense if you consider that mutliple objects of different types can be selected for this tool.

The point UI is already displayed.

https://wiki.freecad.org/Draft_Edit#Supported_objects