FreeCAD/FreeCAD #19651: PartDesign: Object issue when creating a new body (Asm4 related)

leoheck, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am seeing an issue when I am creating a Body (from the Asm4 wb). 

```
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/Assembly4/./newPartCmd.py", line 90, in Activated
    App.ActiveDocument.getObject('YZ_Plane'+index).ViewObject.ShapeColor=(1.0, 0.0, 0.0)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'NoneType' object has no attribute 'ViewObject'"Empty widget item in QVBoxLayout 'verticalLayout_10'."
```

The author said it is an issue on the Freecad itself, see what he said:

> when creating a Part or Body, it creates the Origin object, containing the axes and planes, and creates also a point called also Origin ! Which is therefore renamed Origin001 since 2 objects cannot have the same name. When you then create a new Part or Body, the Origin object is renamed Origin002 (since Origin and Origin001 already exist), while the axes and planes are renamed X_Axis001 since X_Axis already exists. Therefore, the index that is calculated from Origin002 is wrong.

Can someone check if this is something that is off on Freecad, please?

I am using Asm4, and I also have this setting set. Not sure if it is related, tho.

[image]

This issue was reported first here [This is a bug and regression in the FreeCAD development branch, please report it there.](https://github.com/Zolko-123/FreeCAD_Assembly4/issues/534#issuecomment-2660946028)

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-02-16:
I don't think this is an issue of Freecad, I could also push a PR to fix it on Asm4. 
I just want to wait for someone who knows this better to confirm that this is a Freecad regression or something.

For the sake or discussion, here it is my PR, https://github.com/Zolko-123/FreeCAD_Assembly4/pull/535

maxwxyz, 2025-02-16:
@PaddleStroke could you please take a look at the naming of datums?

Syres916, 2025-02-16:
It's quite simple, the change between 1.0 and 1.1dev is the extra Origin object in each Body so the counter used in the existing ASM4 code gets out of sync with the index of the Planes so the [PR](https://github.com/Zolko-123/FreeCAD_Assembly4/pull/535) is required for definite.

leoheck, 2025-02-16:
Thanks @Syres916 sorry to bother you guys. I am closing this issue.

closed 2025-02-16 by leoheck