FreeCAD/FreeCAD #19652: Crash when recomputing part made with previous version of FreeCAD

ii8, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When a file is opened that was created with a previous version of FreeCAD it shows a prompt to recompute: 

> Some document(s) require recomputation for migration purposes. It is highly recommended to perform a recomputation before any modification to avoid compatibility problems.
> Do you want to recompute now?

If I select yes, the program crashes.

It seems this only happens when using Qt6, if FreeCAD is built with Qt5 the crash doesn't happen as far as I can tell.
FreeCAD version is 1.1.0dev.40263 but this also happens with 1.0 release version
Using --safe-mode makes no difference, still crashes.

Example file to reproduce:  [belt-clamp.zip](https://github.com/user-attachments/files/18815499/belt-clamp.zip)  (extension changed from FCStd to zip so github accepts it, just download and change it back to belt-clamp.FCStd)
This happens with all my files, not just this one.

<details>
<summary>Backtrace</summary>

```gdb
Thread 58 "FreeCAD" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 29067]
0x00007fffc76f579c in math_FunctionSetRoot::Perform(math_FunctionSetWithDerivatives&, math_Vector const&, math_Vector const&, math_Vector const&, bool) () from /lib/libTKMath.so.7.7
(gdb) bt
#0  0x00007fffc76f579c in math_FunctionSetRoot::Perform(math_FunctionSetWithDerivatives&, math_Vector const&, math_Vector const&, math_Vector const&, bool) () from /lib/libTKMath.so.7.7
#1  0x00007fffc7c24f53 in Extrema_GenLocateExtPS::Perform(gp_Pnt const&, double, double, bool) () from /lib/libTKGeomBase.so.7.7
#2  0x00007fffc9edf927 in IntTools_BeanFaceIntersector::ComputeRangeFromStartPoint(bool, double, double, double, int) () from /lib/libTKBO.so.7.7
#3  0x00007fffc9ee441a in IntTools_BeanFaceIntersector::ComputeUsingExtremum() () from /lib/libTKBO.so.7.7
#4  0x00007fffc9ee4819 in IntTools_BeanFaceIntersector::Perform() () from /lib/libTKBO.so.7.7
#5  0x00007fffc9efb28a in IntTools_EdgeFace::Perform() () from /lib/libTKBO.so.7.7
#6  0x00007fffc9fd6ead in BOPAlgo_EdgeFace::Perform() () from /lib/libTKBO.so.7.7
#7  0x00007fffc9fd5c09 in OSD_ThreadPool::Job<BOPTools_Parallel::ContextFunctor2<NCollection_Vector<BOPAlgo_EdgeFace>, IntTools_Context> >::Perform(int) () from /lib/libTKBO.so.7.7
#8  0x00007fffc74b1d66 in OSD_ThreadPool::performJob(opencascade::handle<Standard_Failure>&, OSD_ThreadPool::JobInterface*, int) () from /lib/libTKernel.so.7.7
#9  0x00007fffc74b217c in OSD_ThreadPool::EnumeratedThread::performThread() () from /lib/libTKernel.so.7.7
#10 0x00007fffc74b21c9 in OSD_ThreadPool::EnumeratedThread::runThread(void*) () from /lib/libTKernel.so.7.7
#11 0x00007ffff7fb858e in start (p=0x7fffc7301630) at src/thread/pthread_create.c:195
#12 0x00007ffff7fc552b in __clone () at src/thread/x86_64/clone.s:22
Backtrace stopped: frame did not save the PC
```

</details>

### Full version info

```shell
OS: Void Linux (wayland)
Architecture: x86_64
Version: 1.1.0dev.40263 (Git)
Build type: Release
Branch: main
Hash: eb886449c2d63b1b5e61bf6a5b7acac61d2382e5
Python 3.13.2, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/ProDark/fusion
Logical/physical DPI: 96/91.7275
Installed mods: 
  * freecad.gears 1.2.0
  * fasteners 0.5.21
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-02-16:
My Qt6 build does not crash on your file

ii8, 2025-02-16:
Interesting, what version of opencascade do you have?

benj5378, 2025-02-16:
@ii8 can you provide a more extensive backtrace? Would like to see which FreeCAD command that triggers this

ii8, 2025-02-16:
What do you mean by more extensive backtrace? This is the whole backtrace that gdb shows. I can make freecad write this log but I'm not sure it's very useful 

[fc.log](https://github.com/user-attachments/files/18815934/fc.log)

This warning also shows up sometimes but I don't think it's related: "Coin warning in glxglue_isdirect(): Couldn't get current GLX context."

benj5378, 2025-02-16:
@ii8 can you try reproducing the issue with `QT_QPA_PLATFORM=xcb ./bin/FreeCAD`?

ii8, 2025-02-16:
Same thing happens with xwayland

ii8, 2025-02-16:
>so you have tried building both with Qt5 and Qt6 on your own?

Yes that's right

[CMakeConfigureLog.txt](https://github.com/user-attachments/files/18816481/CMakeConfigureLog.txt)

[CMakeConfigureLog-qt5.txt](https://github.com/user-attachments/files/18816482/CMakeConfigureLog-qt5.txt)

benj5378, 2025-02-17:
@ii8 you can turn off building many of the workbenches. After build config, open CMakeCache.txt and turn off some of the workbenches like TechDraw, SpreadSheet and Robot

ii8, 2025-02-17:
I built a FreeCAD with `CMAKE_BUILD_TYPE=Debug`, but the backtrace still looks the same. Maybe the backtrace simply doesn't go further because that's where the thread was started?

I have managed to get gdb to give me this: [gdb.txt](https://github.com/user-attachments/files/18826620/gdb.txt)
Which is backtraces for all the threads instead of just the one where the segfault occurs. The one at the very bottom seems to be the main thread.