FreeCAD/FreeCAD #19654: TechDraw: Crash on selecting 3D edge for dimension

pieterhijma, 2025-02-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD crashes when selecting a 3D edge to create a dimension in TechDraw.

#### To reproduce:

- Create a Part cube.
- Create a TechDraw default page.
- Select the cube in the tree view, do "Insert View", and press OK.
- Go back to the 3D view, select an edge.
- In the tree view, select "Page" -> "View" as well (ctrl click).
- Press "Insert Dimension" in the TechDraw workbench.

#### Expected behavior

At a minimum no crash, but preferably a dimension in the view that contains this edge as 3D reference.

<details>
<summary>Stacktrace:</summary>
<pre>
#0  0x00007fff66c3cacc in TechDrawGui::QGVPage::getDrawPage (this=0x0) at /home/pieter/current/fpa/freecad-main/src/Mod/TechDraw/Gui/QGVPage.cpp:552
#1  0x00007fff66c0d7e0 in TechDrawGui::TechDrawHandler::getPage (this=0x55555f5750a0) at /home/pieter/current/fpa/freecad-main/src/Mod/TechDraw/Gui/TechDrawHandler.cpp:130
#2  0x00007fff66a9734d in TDHandlerDimension::onSelectionChanged (this=0x55555f5750a0, msg=...) at /home/pieter/current/fpa/freecad-main/src/Mod/TechDraw/Gui/CommandCreateDims.cpp:517
#3  0x00007ffff629ec33 in Gui::SelectionObserver::_onSelectionChanged (this=0x55555f5750d8, msg=...) at /home/pieter/current/fpa/freecad-main/src/Gui/Selection/Selection.cpp:153
#4  0x00007ffff62c11a8 in std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&> (__f=@0x55555b40fc50: (void (Gui::SelectionObserver::*)(Gui::SelectionObserver * const, const Gui::SelectionChanges &)) 0x7ffff629ebce <Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&)>, __t=@0x55555b40fc60: 0x55555f5750d8) at /usr/include/c++/14.2.1/bits/invoke.h:74
#5  0x00007ffff62c0a85 in std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&> (__fn=@0x55555b40fc50: (void (Gui::SelectionObserver::*)(Gui::SelectionObserver * const, const Gui::SelectionChanges &)) 0x7ffff629ebce <Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&)>) at /usr/include/c++/14.2.1/bits/invoke.h:96
#6  0x00007ffff62c018a in std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>(std::tuple<Gui::SelectionChanges const&>&&, std::_Index_tuple<0ul, 1ul>) (this=0x55555b40fc50, __args=...) at /usr/include/c++/14.2.1/functional:513
#7  0x00007ffff62bf74a in std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::operator()<Gui::SelectionChanges const&, void>(Gui::SelectionChanges const&) (this=0x55555b40fc50) at /usr/include/c++/14.2.1/functional:598
#8  0x00007ffff62beba0 in boost::detail::function::void_function_obj_invoker<std::_Bind<void (Gui::SelectionObserver::*(Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) (function_obj_ptr=..., a#0=...) at /usr/include/boost/function/function_template.hpp:100
#9  0x00007ffff5dd9b10 in boost::function_n<void, Gui::SelectionChanges const&>::operator() (this=0x55555b40fc48, a#0=...) at /usr/include/boost/function/function_template.hpp:789
#10 0x00007ffff5dd6d44 in boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::m_invoke<boost::function<void(Gui::SelectionChanges const&)>, 0u, Gui::SelectionChanges const&> (this=0x7fffffff9c17, func=..., args=std::tuple containing = {...}) at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:105
#11 0x00007ffff5dd29e9 in boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::operator()<boost::function<void(Gui::SelectionChanges const&)>, Gui::SelectionChanges const&, 1ul> (this=0x7fffffff9c17, func=..., args=std::tuple containing = {...}) at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:90
#12 0x00007ffff5dce593 in boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>::operator()<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > > (this=0x7fffffff9f50, connectionBody=...) at /usr/include/boost/signals2/detail/variadic_slot_invoker.hpp:133
#13 0x00007ffff5dc9288 in boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> >::dereference (this=0x7fffffff9d00) at /usr/include/boost/signals2/detail/slot_call_iterator.hpp:108
#14 0x00007ffff5dc38cc in boost::iterators::iterator_core_access::dereference<boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > > (f=...) at /usr/include/boost/iterator/iterator_facade.hpp:631
#15 0x00007ffff5dbddd4 in boost::iterators::detail::iterator_facade_base<boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> >, boost::signals2::detail::void_type, boost::iterators::single_pass_traversal_tag, boost::signals2::detail::void_type&, long, false, false>::operator* (this=0x7fffffff9d00) at /usr/include/boost/iterator/iterator_facade.hpp:737
#16 0x00007ffff5db5fef in boost::signals2::optional_last_value<void>::operator()<boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void(Gui::SelectionChanges const&), boost::function<void(Gui::SelectionChanges const&)> >, boost::signals2::mutex> > > (this=0x555555ded450, first=..., last=...) at /usr/include/boost/signals2/optional_last_value.hpp:58
#17 0x00007ffff5dacab2 in boost::signals2::detail::combiner_invoker<void>::operator()<boost::signals2::optional_last_value<void>, boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_sl [... cut ...]
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical/physical DPI: 96/92.6073
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

### Subproject(s) affected?

TechDraw

### Anything else?

I'm trying to insert a dimension as described on [this wiki](https://wiki.freecad.org/TechDraw_LengthDimension#Usage).

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-16:
did it work in 1.0?

pieterhijma, 2025-02-16:
Good question. No, it crashes 1.0 as well, so no regression.

maxwxyz, 2025-02-16:
I cannot reproduce, it adds the dimension as expected. Could you share your file?

```
OS: macOS 13.7.3
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/72
Installed mods: 
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * CfdOF 1.29.8
  * Defeaturing 1.2.3
  * Manipulator 1.5.7
  * fasteners 0.5.37
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.11
  * Curves 0.6.58

```

pieterhijma, 2025-02-16:
> I cannot reproduce, it adds the dimension as expected. Could you share your file and details on which view(s) you have created, which edges and view you have selected?

Thanks, I verified on the same Mac OS weekly build as well (but ARM) and indeed it works there more or less as expected.
I also verified on Windows 40239 x86_64 and there it works as expected as well.

However, not all is good. On Mac OS I can cause a crash by making a slight modification to the steps to reproduce:

- Create a Part cube.
- Create a TechDraw default page.
- Select the cube in the tree view, do "Insert View", and press OK.
- Go back to the 3D view, select an edge.
- **Changed wrt above, don't do this step:** In the tree view, select "Page" -> "View" as well (ctrl click).
- Press "Insert Dimension" in the TechDraw workbench.

On Mac OS this causes a crash. On Windows I get the following exceptions repeatedly:

```
<Exception> Access violation
0 Selection.cpp(156): Unhandled Base::Exception caught in selection observer:
```

pieterhijma, 2025-02-16:
> BTW what is the benefit of creating dimensions this tortuous way, instead of adding them on the page?

:) The benefit of this tortuous way is that you immediately get a reference to the 3D edge besides the 2D reference. For example, if you have a view in perspective (for example the camera icon) and you would add a dimension in this way, it would show you the projected length initially which is not 10 mm anymore because the line is diagonal. However, since the actual 3D edge is known, you can set the `MeasureType` to `True` to get the true length instead of the projected length. 

In the less tortuous way, this is not possible with simply selecting the edge in the 2D view because there is no relation to the original 3D edge that is aware of the true edge length.

benj5378, 2025-02-16:
I can reproduce. Seems like the segfault only occurs upon switching to the TechDraw Page tab. Also seems like the segfault is somewhat weirdly delayed.

WandererFan, 2025-02-17:
Putting this here so I don't lose it in my files.

My guess is that the viewPage variable in TechDrawHandler/TechDrawHandlerDimension is nullptr when it calls QGVPage::getDrawPage().  

Not familiar at all with this handler stuff that paddle added.

```
#0  0x00007fff66c3cacc in TechDrawGui::QGVPage::getDrawPage (this=0x0) at /home/pieter/current/fpa/freecad-main/src/Mod/TechDraw/Gui/QGVPage.cpp:552
#1  0x00007fff66c0d7e0 in TechDrawGui::TechDrawHandler::getPage (this=0x55555f5750a0) at /home/pieter/current/fpa/freecad-main/src/Mod/TechDraw/Gui/TechDrawHandler.cpp:130
#2  0x00007fff66a9734d in TDHandlerDimension::onSelectionChanged (this=0x55555f5750a0, msg=...) at /home/pieter/current/fpa/freecad-main/src/Mod/TechDraw/Gui/CommandCreateDims.cpp:517

CommandCreateDims.cpp:516 ...
        if (msg.Object.getObjectName().empty()
            || msg.Object.getDocument() != getPage()->getDocument()) {  <<< what if getPage returns nullptr?
            if (msg.Type == Gui::SelectionChanges::AddSelection) {
                Gui::Selection().rmvSelection(msg.pDocName, msg.pObjectName, msg.pSubName);
            }
            return;
        }
        

TechDrawHandler.cpp:128 ...
TechDraw::DrawPage* TechDrawHandler::getPage()
{
    return viewPage->getDrawPage();         // viewPage is a QGVPage(QGraphicsView) widget
}                                           // viewPage is nullptr? (this=0x0)

QGVPage.cpp:552
TechDraw::DrawPage* QGVPage::getDrawPage() { return m_vpPage->getDrawPage(); }

```

WandererFan, 2025-02-17:
```
void activateHandler(TechDrawHandler* newHandler)
{
    auto* mdi = dynamic_cast<MDIViewPage*>(Gui::getMainWindow()->activeWindow());   <<< if selecting a 3d edge, the activeWindow will never be
                                                                                    <<< the desired DrawPage.  activateHandler will never be called
                                                                                    <<< and viewPage will never get set correctly.
    if (!mdi) {
        return;
    }
...
   viewPage->activateHandler(newHandler);
}
```

WandererFan, 2025-02-17:
@benj5378, @pieterhijma when you reproduce the error are you using the single tool or the separate tools.  I use the separate tools - that might be why I get different results.

If I use the combined tool, the application exits shortly after making a dimension.

pieterhijma, 2025-02-17:
@WandererFan: Right, I'm doing "Insert Dimension", command `TechDraw_Dimension`. If I do "Insert Length Dimension", command `TechDraw_LengthDimension`, then it works as expected.

pieterhijma, 2025-02-17:
> On linux mint 21.2:
> 
> If I follow the mac instructions, I get this every time: [image]

I have this outcome as well when I do "Insert Length Dimension". So, the problem is with "Insert Dimension", command `TechDraw_LengthDimension`.

WandererFan, 2025-02-19:
This is actually an instance of #14238.

maxwxyz, 2025-02-19:
That's why I could not reproduce.

closed 2025-02-19 by maxwxyz

pieterhijma, 2025-02-19:
Ah right, sorry, I missed that issue.